huggingface/lerobot #547: How to make a custom dataset with v2?

alik-git, 2024-12-04:
Hi guys, thanks for the amazing work.

I am trying to make a custom dataset to use with the LeRobotDataset format.

The readme says

alik-git, 2024-12-04:
okay so I've found a work around for now. I initialize an empty dataset and add the frames to it, and then I can load it after calling dataset.consolidate(). If this is a proper way to do it, pls lmk and I'll make a PR with updates to the docs. 

Otherwise please let me know what the right way to do this is. Thank you! I'll update this with my code once I've cleaned it up.

Robert-hua, 2024-12-05:
I encountered the same issue.

taochenshh, 2024-12-08:
@aliberts i also got the same issue, the documentation on how to generate custom dataset is not up to date now (the code doesn't run anymore). could you please up the instruction and relevant scripts for custom dataset generation? thanks

aliberts, 2024-12-11:
Hey there,
Yes, all the `push_to_hub` script are deprecated in favor of scripts in [`examples/port_datasets`](https://github.com/huggingface/lerobot/tree/main/examples/port_datasets) (just one for now).

Basically, you need to create a new empty dataset using `LeRobotDataset.create()`, then add individual frames using `add_frame()`, then save the added frames into an episode using `save_episode()` (which actually saves data).
Then at the end you need to call the `consolidate()` method to handle a few more things (we will try to get rid of this step in future iterations) before finally calling the `push_to_hub()` method.

You can find more info about the changes of this new api in the PR ([#461](https://github.com/huggingface/lerobot/pull/461))

We will remove `push_to_hub.py` scripts in the future after adding more equivalent scripts like the one mentioned above in the examples section. Hope this helps!

aliberts, 2024-12-11:
Will update the Readme soon!

tlpss, 2025-02-07:
@aliberts, I stumbled upon this problem and could not find it in the readme? Could you make it more explicit?

AbdElrahmanMostafaRifaat1432, 2025-02-08:
can anybody give us the code to to do that with a static aloha dataset format

Wonder1905, 2025-02-09:
I've followed:
https://github.com/huggingface/lerobot/blob/main/examples/port_datasets/pusht_zarr.py

But this is extremely memory intensive, using every memory trick I know didnt help it to finish single episode.
Maybe I dont understand whats going "under the hood" but holding a dataset object, that keeps getting multiple frames seems a bit crazy. But maybe Im missing something and I'll be happy for some help here.

tlpss, 2025-02-09:
@AbdElrahmanMostafaRifaat1432, I have not used the aloha format but I created a [script](https://github.com/tlpss/lerobot/blob/dp_zarr_dataset/examples/port_datasets/dp_3d_zarr.py)  to convert datasets using the format from the diffusion policy codebase. THe process for the aloha format will be very similar, I believe.

AbdElrahmanMostafaRifaat1432, 2025-02-09:
@tlpss 
I really appreciate your response but I have found an easier solution, I took the script used by openpi to push aloha dataset   

you can find it here: [convert_aloha_to_lerobot](https://github.com/PhysicalIntelligence/openpi/blob/main/examples/aloha_real/convert_aloha_data_to_lerobot.py)

I wonder if there is a solution to convert data immediately locally instead of pushing it to hugging face and taking it again from hugging face.
this would help me a lot actually

tlpss, 2025-02-09:
@AbdElrahmanMostafaRifaat1432, that's even easier! 

> I wonder if there is a solution to convert data immediately locally instead of pushing it to hugging face and taking it again from hugging face.

If you set [this](https://github.com/Physical-Intelligence/openpi/blob/80d346ea0d64fd92f663c705b687bd5fc4bb96a0/examples/aloha_real/convert_aloha_data_to_lerobot.py#L236)  variable to False, the dataset will only be stored locally.  Don't forget to set the `LEROBOT_HOME` env variable to the dir where you want to store the dataset.

AbdElrahmanMostafaRifaat1432, 2025-02-10:
actually the problem is when I train the model it need to see the repo otherwise it will produce error that it cannot find the repo

when I use the following script I have error
python lerobot/scripts/train.py --policy.type=pi0 --dataset.repo_id=Abdorifaat/basket

Tavish9, 2025-02-19:
Hi all, I have release a script about the conversion from OpenX(RLDS) to LeRobotV2.0, see [this](https://github.com/Tavish9/openx2lerobot).

You can easily modify the code to make a custom LeRobotDataset, by just changing the source and everything is done. 🥳

closed 2025-10-08 by pkooij