huggingface/lerobot #552: Rounding to int32 makes robot less precise. Do we have a solid reason for doing this?

1g0rrr, 2024-12-05:
### System Info

```Shell
Latest LeRobot. MacOS
```

### Information

- [X] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

1) Run teleoperation
2) Measure preciseness with rounding and without.
at lerobot/common/robot_devices/robots/manipulator.py

[image]

### Expected behavior

Smooth movement

apockill, 2024-12-28:
I also noticed this when I was trying using radians for my robot arm.

vaishanth-rmrj, 2025-01-30:
I tried it and I didn't notice any significant change in the robot motion smoothness.

apockill, 2025-01-31:
I do still think we shouldn't do the conversion- specifically for cases where the unit isn't in degrees, or if your robot does require precision. In radians it's unusable unless you remove the conversion.

closed 2025-10-08 by jadechoghari