huggingface/lerobot #568: Do I need two SO 100 arms to get started?

rabhishek100, 2024-12-10:
I have printed and assembled one arms, the follower version. Do I need two arms to record datasets and do testing?

JohnsonLv, 2024-12-10:
you need a leader to tell follower what to do

rabhishek100, 2024-12-10:
can i use the same arm as the leader to record datasets and as a follower while running policies?

fracapuano, 2024-12-30:
Hey @rabhishek100 👋 

## Short answer
Nope

## Long answer
- Assuming you are training policies using imitation learning directly from frames (that is, using ACT and thus following the tutorial quite verbatim),
- Assuming you intend to directly manouver the sole (follower) arm while recording

then you would be in the camera frames when imitation-learning, which would make learning impractical. You could be using one arm only in the case where you were demonstrating manouvering the robotic arm, but you would have to learn from sensors (in the case of the arm, joint angles + your own sensors), which I think is a bit beyond the current scope of the arm. For that, I think you would be much better off 3D-printing the other arm and using _teleoperation_ to record your dataset instead of _manouvering_.

In case you are not learning from pixels, but rather from sensors, hmu! I would love to learn more about what you are doing and perhaps help you if needed 😊

rabhishek100, 2024-12-31:
Hi @fracapuano, thanks for the response.

I understand what you mean. We cannot have our hands in the frames while learning. When I started off with this, I had a few free servos laying around and I was able to assemble an arm. I was thinking of a way to do imitation learning with just the one arm. I played around with the arm and was able to teleoperate it pretty well using just a keyboard, by doing inverse kinematics. But later, I got around to building a second arm and successfully trained act for a simple pick and place task. It would be great if we can use keyboard teleop for training the arm (we can read the actions directly from the follower arm itself). This way, people can easily get started with just a single arm, reducing the time and cost by half. Let me know if I am missing something here.

This is the keyboard teleop I was able to achieve - https://www.youtube.com/shorts/emt-0YKsKAA

fracapuano, 2024-12-31:
@rabhishek100 great! Would you mind putting here the inverse kinematics code you have produced? I would gladly give it a look 👍

rabhishek100, 2024-12-31:
Sure! I have attached the files
[keyboard_teleop.zip](https://github.com/user-attachments/files/18282214/keyboard_teleop.zip)

YanJiaHuan, 2025-02-05:
Hi, everyone, i am looking for which camera they use for this robot, but it seems they just use webcamera. It is cool but the laoptop and phone will not always be located in same place, i believe the ideal setup is one or two fixed cameras either bird eye view or attached on wrist. 
Did they claim where to setup this fixed cameras? Or is it hard if i want to use any general cameras like realsense or hkvision?

sourcesync, 2025-04-27:
Hi @rabhishek100 were you able to do imitation learning with just one arm and keyboard kinematics (what you described above?). Thanks!

ramallis, 2025-06-27:
has anyone made a calibration file for so 100 robot arm that only has the follower

fracapuano, 2025-06-28:
Hey @ramallis, our calibration setup as of now provides support for separate calibration of both leader and follower. Isn't that what you see too?

ramallis, 2025-06-29:
I did use the original file for calibration, but the detection of the ID does not work for the motors 2-6, although it works for the first one. The command I am doing to set the ID of each one is 'python lerobot/scripts/configure_motor.py --port /dev/ttyACM0 --brand feetech --model sts3215 --baudrate 1000000. Now, I have tried the other two recommended IDs (57600 and 115200, but the ID still does not work for the remaining motors). How should I proceed? Thank you.

fracapuano, 2025-10-08:
wow @BrosnanYuen this is cool 🤯 you may also want to check out the new phone teleop system, which allows you to entirely skip dealing with a leader arm anyway and has slightly faster reaction times (no need to run rlt inference on images and so on). I would recommend trying it out!