huggingface/lerobot #596: How to achieve multiple tasks on the basis of LeRobot ？

wangwisdom, 2024-12-25:
LeRobot can achieve single tasks (such as inserting, transferring blocks, etc.), how to achieve multiple tasks on the basis of LeRobot (such as first recognizing objects and classifying, and then putting objects in order in boxes, etc.)?"
Please give me some ideas.

Vacuame, 2024-12-25:
There is a class called "MultiLeRobotDataset", but i dont't know how to use it

TangGuohh, 2025-04-01:
Hello, I also struggled with this issue for a long time, but I've resolved it. The solution is efficient and verified to work correctly. You can use the merge.py script from my PR (#924) to solve your problem. I hope this helps you, and wish you all the best.