huggingface/lerobot #607: maximum range of [-10, 110] % to account for some imprecision during calibration, but present value is inf %

mbauer599, 2024-12-31:
### System Info

```Shell
- `lerobot` version: 0.1.0
- Platform: Linux-5.15.0-122-generic-x86_64-with-glibc2.35
- Python version: 3.10.12
- Huggingface_hub version: 0.27.0
- Dataset version: 3.2.0
- Numpy version: 2.0.2
- PyTorch version (GPU?): 2.5.1+cu124 (False)
- Cuda version: 12040

I'm using a SO-100 arm that is almost 100% stock, with the exception that they sent me 12v motors which works just fine for me (or should).
```

### Information

- [ ] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

```
~/lerobot$ python3.10 lerobot/scripts/control_robot.py teleoperate --robot-path lerobot/configs/robot/so100.yaml
Connecting main follower arm.
Connecting main leader arm.
Activating torque on main follower arm.
/home/mbauer/lerobot/lerobot/common/robot_devices/motors/feetech.py:471: RuntimeWarning: divide by zero encountered in scalar divide
  values[i] = (values[i] - start_pos) / (end_pos - start_pos) * 100
Wrong motor position range detected for gripper. Expected to be in nominal range of [0, 100] % (a full linear translation), with a maximum range of [-10, 110] % to account for some imprecision during calibration, but present value is inf %. This might be due to a cable connection issue creating an artificial jump in motor values. You need to recalibrate by running: `python lerobot/scripts/control_robot.py calibrate`
/home/mbauer/lerobot/lerobot/common/robot_devices/motors/feetech.py:543: RuntimeWarning: divide by zero encountered in scalar divide
  calib_val = (values[i] - start_pos) / (end_pos - start_pos) * 100
Traceback (most recent call last):
  File "/home/mbauer/lerobot/lerobot/common/robot_devices/motors/feetech.py", line 404, in apply_calibration_autocorrect
    values = self.apply_calibration(values, motor_names)
  File "/home/mbauer/lerobot/lerobot/common/robot_devices/motors/feetech.py", line 474, in apply_calibration
    raise JointOutOfRangeError(
lerobot.common.robot_devices.motors.feetech.JointOutOfRangeError: Wrong motor position range detected for gripper. Expected to be in nominal range of [0, 100] % (a full linear translation), with a maximum range of [-10, 110] % to account for some imprecision during calibration, but present value is inf %. This might be due to a cable connection issue creating an artificial jump in motor values. You need to recalibrate by running: `python lerobot/scripts/control_robot.py calibrate`

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/mbauer/lerobot/lerobot/scripts/control_robot.py", line 564, in <module>
    teleoperate(robot, **kwargs)
  File "/home/mbauer/lerobot/lerobot/common/robot_devices/utils.py", line 28, in wrapper
    raise e
  File "/home/mbauer/lerobot/lerobot/common/robot_devices/utils.py", line 24, in wrapper
    return func(robot, *args, **kwargs)
  File "/home/mbauer/lerobot/lerobot/scripts/control_robot.py", line 177, in teleoperate
    control_loop(
  File "/home/mbauer/lerobot/lerobot/common/datasets/image_writer.py", line 36, in wrapper
    raise e
  File "/home/mbauer/lerobot/lerobot/common/datasets/image_writer.py", line 29, in wrapper
    return func(*args, **kwargs)
  File "/home/mbauer/lerobot/lerobot/common/robot_devices/control_utils.py", line 242, in control_loop
    robot.connect()
  File "/home/mbauer/lerobot/lerobot/common/robot_devices/robots/manipulator.py", line 343, in connect
    self.follower_arms[name].read("Present_Position")
  File "/home/mbauer/lerobot/lerobot/common/robot_devices/motors/feetech.py", line 760, in read
    values = self.apply_calibration_autocorrect(values, motor_names)
  File "/home/mbauer/lerobot/lerobot/common/robot_devices/motors/feetech.py", line 407, in apply_calibration_autocorrect
    self.autocorrect_calibration(values, motor_names)
  File "/home/mbauer/lerobot/lerobot/common/robot_devices/motors/feetech.py", line 565, in autocorrect_calibration
    raise ValueError(f"No integer found between bounds [{low_factor=}, {upp_factor=}]")
ValueError: No integer found between bounds [low_factor=np.float32(-0.0007324219), upp_factor=np.float32(-0.0007324219)]
```

It looks like all the calibration data is generated without issue.
```
~/lerobot$ cat .cache/calibration/so100/main_follower.json 
{"homing_offset": [4658, -2773, 645, -2757, -2358, -3022], "drive_mode": [1, 0, 0, 0, 0, 0], "start_pos": [3635, 3797, 379, 3781, 3382, 4046], "end_pos": [-3634, 3797, 379, 3781, 3382, 4046], "calib_mode": ["DEGREE", "DEGREE", "DEGREE", "DEGREE", "DEGREE", "LINEAR"], "motor_names": ["shoulder_pan", "shoulder_lift", "elbow_flex", "wrist_flex", "wrist_roll", "gripper"]}
```
```
~/lerobot$ cat .cache/calibration/so100/main_leader.json 
{"homing_offset": [-2555, 628, 205, 655, -878, 977], "drive_mode": [0, 0, 0, 0, 0, 0], "start_pos": [3579, 396, 819, 369, 1902, 46], "end_pos": [3579, 396, 819, 369, 1902, 47], "calib_mode": ["DEGREE", "DEGREE", "DEGREE", "DEGREE", "DEGREE", "LINEAR"], "motor_names": ["shoulder_pan", "shoulder_lift", "elbow_flex", "wrist_flex", "wrist_roll", "gripper"]}
```

This also occurs when attempting to execute teleoperate or any other control mode. 

### Expected behavior

Execution without error. 

If this is a 'me' issue, then apologies in advanced for not just throwing this on discord, but any help or guidance is appreciated.

abhijitmajumdar, 2025-01-20:
I have the same setup as you do and was running into a similar issue with `JointOutOfRangeError` despite calibration. Turns out I had my motor IDs reversed when I was configuring the motors. A simple change in the `lerobot/configs/robot/so100.yaml` fixed it. My `so100.yaml` looks so this now
```yaml
leader_arms:
  main:
    _target_: lerobot.common.robot_devices.motors.feetech.FeetechMotorsBus
    port: /dev/ttyACM1
    motors:
      # name: (index, model)
      shoulder_pan: [6, "sts3215"]
      shoulder_lift: [5, "sts3215"]
      elbow_flex: [4, "sts3215"]
      wrist_flex: [3, "sts3215"]
      wrist_roll: [2, "sts3215"]
      gripper: [1, "sts3215"]

follower_arms:
  main:
    _target_: lerobot.common.robot_devices.motors.feetech.FeetechMotorsBus
    port: /dev/ttyACM0
    motors:
      # name: (index, model)
      shoulder_pan: [6, "sts3215"]
      shoulder_lift: [5, "sts3215"]
      elbow_flex: [4, "sts3215"]
      wrist_flex: [3, "sts3215"]
      wrist_roll: [2, "sts3215"]
      gripper: [1, "sts3215"]
```

For me though the the error explicitly mentioned the "present position" to be -10.1667, which was a hint to it being out of range when it should not, unlike your infinity. So the error you are seeing might be because of something else, if not the motor IDs

mbauer599, 2025-01-22:
In this case the error was in my face, and I did have the servo positioned incorrectly, but I think this is still more of a math issue than anything. I did reorient the motors and everything works good now. 
```
lerobot.common.robot_devices.motors.feetech.JointOutOfRangeError: Wrong motor position range detected for gripper.
```

Davegdd, 2025-01-30:
Leaving this here in case it helps someone.
Had this error non-stop but with present position 50550 or so and this is what I figured out and how I solved it. 
I kept getting the error when trying to calibrate follower arm doing:

```
python lerobot/scripts/control_robot.py calibrate \
    --robot-path lerobot/configs/robot/so100.yaml \
    --robot-overrides '~cameras' --arms main_follower
```

But it turned out the follower was perfectly fine and the problem was in the leader, I got confused given the previous command and since the output on the terminal did not specify which arm failed. To solve this you have to correctly adjust the position of the servos. For this just download any version of [Feetech debugging software](https://www.feetechrc.com/software.html) (I used FD1.9.8.3 on a Windows PC), choose the COM port your servos are connected to, select Baud Rate 1000000, click Open, Click Search, you should see your 6 servos. Then, select them one by one and in the Programming tab enter 128 in Address 40:

[image]

klhftco, 2025-02-26:
Leaving as a data point. 

Also had this error, turns out the gripper servo was capturing the same value between rotated and zero positions, resulting in the start and end position being the same and inf value for present value (by 0 division), so make sure that between the two calibration positions, the gripper is put in the lower and upper bounds of its limits, so that the positions are read correctly. 

It also helps to set some pdb traces in `feetech_calibration.py` and `control_robot.py` just to follow along with the calibration logic, and verify the positions that are being read make sense.

gxy-TJU, 2025-04-05:
> Leaving as a data point.
>
> Also had this error, turns out the gripper servo was capturing the same value between rotated and zero positions, resulting in the start and end position being the same and [...]

I seem have same problem as you, how do you correct it?

klhftco, 2025-04-05:
> > Leaving as a data point.
> > Also had this error, turns out the gripper servo was capturing the same value between rotated and zero positions, resulting in the start and end position being the same and inf value for present value (by 0 division), so make sure that between the two calibration positions, the gripper is put in the lower and upper bounds of its limits, so that the positions are read correctly.
> > It also helps to set some pdb traces in `feetech_calibration.py` and `control_robot.py` just to follow along with the calibration logic, and verify the positions that are being read make sense.
> 
> I seem have same problem as you, how do you correct it? Thanks for any response!

Yeah, between calibration positions for the leader arm, just make sure you're holding the position a little longer so it catches the correct position, and you can also print the position after each calibration to make sure they're different. If all else fails, you can also just manually set the offsets in the config file that it writes to, by finding the difference between the zero and rotated positions.

Marius-Ar, 2025-04-18:
Same issue here.

Tried :
- calibrating tens of times but it still prints errors stating that a motor is in an invalid position, a different one almost each time.
- removing content from `.cache/calibration/so100`.
- checking all 6 servos using Feetech servos debugger software.

I only have a follower arm and can't really say that this configuration is well documented yet.

amrltqt, 2025-04-25:
Got this issue today, I realized that it was due to my poor calibration of the 6 servos. 
It's mandatory to respect exactly the position of EACH servos following this guide https://github.com/huggingface/lerobot/blob/main/examples/10_use_so100.md#a-manual-calibration-of-follower-arm

Once i replay the calibration by respecting the 6 positions, I got a good calibration for both arms.

Lottie56, 2025-05-15:
> Leaving this here in case it helps someone. Had this error non-stop but with present position 50550 or so and this is what I figured out and how I solved it. I kept getting the error when trying to calibrate follower arm doing:
> 
> ```
> python lerobot/scripts/control_robot.py calibrate \
>     --robot-path lerobot/configs/robot/so100.yaml \
>     --robot-overrides '~cameras' --arms main_follower
> ```
> 
> But it turned out the follower was perfectly fine and the problem was in the leader, I got confused given the previous command and since the output on the terminal did not specify which arm failed. To solve this you have to correctly adjust the position of the servos. For this just download any version of [Feetech debugging software](https://www.feetechrc.com/software.html) (I used FD1.9.8.3 on a Windows PC), choose the COM port your servos are connected to, select Baud Rate 1000000, click Open, Click Search, you should see your 6 servos. Then, while holding each part of the arm more or less on the middle of its range (as per the building video), select the servos one by one and in the Programming tab enter 128 in Address 40:
> 
> [image]

What is the purpose to set torque enable to 128 ?

closed 2025-10-08 by pkooij