huggingface/lerobot #613: Starting off with pretrained models

rabhishek100, 2025-01-03:
Are there any pretrained models available that can be fine tuned using our own dataset for tasks like pick and place and manipulation?

aliberts, 2025-01-09:
Currently it's not super easy to fine-tune pretrained policies due to several limitations in the configs design. We're addressing this in #550 and we'll keep improving usability afterwards.
In the meantime, you can have a look at [`examples/2_evaluate_pretrained_policy.py`](https://github.com/huggingface/lerobot/blob/main/examples/2_evaluate_pretrained_policy.py) to get an idea of how to use (in eval) a pretrained policy but keep in mind it's going to obsolete very soon. Our pretrained policies are available on https://huggingface.co/lerobot but we'll have to retrain them with the current redesign.

rabhishek100, 2025-01-09:
Got it. Thanks!