huggingface/lerobot #624: Diffusion Policy: Transformer backbone support

YanJiaHuan, 2025-01-09:
Thanks for your sharing.
My request is like:
They tried a lost of backbones in original Diffusion Policy, which includes the transformer. Even in their paper, they claimed that the transformer seems to be not as good and stable as Unet backbone, but from my previous reproduce and others work, it is not quite true to claim transformer is worse (no matter the number of params).
Hence, I suggest to add the transformer support in lerobot project. As an advice: You can add a Class named DiffusionTransformer in modeling_diffusion.py and let the user to select the backbone in configuration file.

mohitydv09, 2025-01-23:
This is  PR #481 for this, you can track this: https://github.com/huggingface/lerobot/pull/481

jadechoghari, 2025-10-08:
PR opened, closing the issue: https://github.com/huggingface/lerobot/pull/481

closed 2025-10-08 by jadechoghari