huggingface/lerobot #633: [Question] How to set training to a local dataset?

tlpss, 2025-01-13:
Is there a way to train on a local dataset without manually adding the `local_files_only` arg to the `make_dataset` function of the train script?

I have set the `LEROBOT_HOME` env variable.

aliberts, 2025-01-30:
We plan to have a dataset v2.1 update in the coming weeks, we will adress this. The default behavior should be the same as what was have with `HubMixin` in the upcoming refactor:
- Check if whatever path that was given corresponds to a local dir first
- if not, try to load from the hub

oym1994, 2025-03-07:
> We plan to have a dataset v2.1 update in the coming weeks, we will adress this. The default behavior should be the same as what was have with `HubMixin` in the upcoming refactor:
> 
> * Check if whatever path that was given corresponds to a local dir first
> * if not, try to load from the hub

Hello, have u finished creating this feature? 
I have downloaded the dataset to the local, but every time it will try to download from the hub. 
How can we set to use local dataset for training?

david-knigge, 2025-07-24:
Any news on this or possible solutions? I'm running into the same issue currently

jadechoghari, 2025-10-08:
We removed the use of `local_files_only` https://github.com/huggingface/lerobot/blob/main/src/lerobot/configs/train.py#L36

You can now just pass `dataset.root` specifiying the location of your data