huggingface/lerobot #671: Using cameras adds a 20s delay to teleoperation start time on windows

Joeclinton1, 2025-02-01:
### System Info

```Shell
Windows 11
```

### Information

- [x] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

// Koch - with camera
python lerobot/scripts/control_robot.py teleoperate --robot-path lerobot/configs/robot/koch.yaml

// Koch - with cam and profiling
python -m cProfile -o profile_output.prof lerobot/scripts/control_robot.py teleoperate --robot-path lerobot/configs/robot/koch.yaml

[image]

If on windows you should see this image, when running the cprofile version with a suspiciously large image_writer cell. This is the problematic part that delays the teleoperation.

### Expected behavior

Teleoperation should be able to start in under 2s. There is an issue with the backend being chosen, that causes using the camera to be quite slow. I'll try to put out a pr soon to fix this.

closed 2025-03-08 by aliberts

LaurentBerger, 2025-04-21:
To disable delay May be you can try to set Environment flag ["OPENCV_VIDEOIO_MSMF_ENABLE_HW_TRANSFORMS" set to 0](https://docs.opencv.org/4.x/d4/d15/group__videoio__flags__base.html#ga023786be1ee68a9105bf2e48c700294d)

Joeclinton1, 2025-04-21:
> To disable delay May be you can try to set Environment flag ["OPENCV_VIDEOIO_MSMF_ENABLE_HW_TRANSFORMS" set to 0](https://docs.opencv.org/4.x/d4/d15/group__videoio__flags__base.html#ga023786be1ee68a9105bf2e48c700294d)

Oh we merged my fix that changes backend to solve problem. Is it possible we've simply chosen a backend that doesn't support hardware transforms and hence slowed overall teleoperation?

PR #676 (2025-03-08): Fix delay in teleoperation start time
## What this does

Fixes #671 which is an issue where there is a delay that occurs on windows when teleoperating that forces the user to wait 20s before they gain control of their robot. It reduces this delay to 2s by choosing a better opencv camera backend when on windows.

## How to checkout & try? (for the reviewer)

```bash
python -m cProfile -o profile_output.prof lerobot/scripts/control_robot.py teleoperate --robot-path lerobot/configs/robot/koch.yaml
````
run the above before and after the change to observe the difference in delay. This command uses [... cut]