huggingface/lerobot #674: Can't evaluate the pretrained model due to draccus parser.

Kimho666, 2025-02-02:
[image]

I try example_2 to evaluate pretrained model but error occurs.

aliberts, 2025-02-02:
We didn't push the pretrained models on the hub yet so it's pulling the old versions which don't work with this new config system. We'll update them very soon

ylyhlh, 2025-02-03:
getting the same error.

aliberts, 2025-02-03:
I've just pushed all pretrained policy on https://huggingface.co/lerobot (except `diffusion_pusht_keypoints`).
Example should be working now.

closed 2025-02-03 by aliberts

Kimho666, 2025-02-03:
Yes, the models on hugging are updated 3 hours ago.

ankile, 2025-05-12:
Hi, I'm still getting this issue when trying to load a diffusion policy that was created with the "official" training script: 

```
python lerobot/scripts/train.py \                                          
    --dataset.repo_id=ankile/franka-lift \
    --policy.type=diffusion \             
    --wandb.enable=true
```

When calling `policy = DiffusionPolicy.from_pretrained(pretrained_policy_path)` with the resulting saved models, I get

```
draccus.utils.ParsingError: Expected a dict with a 'type' key for <class 'lerobot.configs.policies.PreTrainedConfig'>, got {'n_obs_steps': 2,....
```

which seems strange to me because the repo is freshly installed as of a couple of days ago and using the standard training script and standard DP config.

Any hints?

shantanuparab-tr, 2025-05-12:
Hey this is not the perfect solution.

```json
    "type": "act",
    "n_obs_steps": 1,
    "normalization_mapping": {
        "VISUAL": "MEAN_STD",
        "STATE": "MEAN_STD",
        "ACTION": "MEAN_STD"
    },
```

Adding a `type` key in your `config.json` file of the policy might solve it if it already does not exists. In your case type should be diffusion.