huggingface/lerobot #678: The inverse kinematic solution code of so-100

gxy-TJU, 2025-02-04:
Are there any code of inverse kinematic of so-100, which just need the input of the x, y on my desk, then it can move to the target 
coordinate？
Thanks for any response.

helper2424, 2025-02-04:
We were discussing it, but I think that nobody started developing it. It should be pretty - just like 2/3 lines of code.

The only important note is that during the building leader arm for SO-100 we remove on gear, so the instruction should adapted to have inverse kinematics.

Davegdd, 2025-03-05:
I second this, supporting a traditional control method would be superuseful. I tried but could not make it work with my arm.

satvikahuja, 2025-03-30:
+++

leo-berte, 2025-04-19:
Hi! I noticed this issue and wanted to share that I’ve implemented inverse kinematics support, including:

- Forward kinematics via DH parameters.
- Analytical Jacobian computation.
- Inverse kinematics using Damped Least Squares (DLS) to avoid singularities.
- Pose interpolation: linear (position) + SLERP (orientation)

I’ll submit a pull request to contribute this module to the repository!

matbeedotcom, 2025-08-11:
> Hi! I noticed this issue and wanted to share that I’ve implemented inverse kinematics support, including:
> 
> * Forward kinematics via DH parameters.
> * Analytical Jacobian computation.
> * Inverse kinematics using Damped Least Squares (DLS) to avoid singularities.
> * Pose interpolation: linear (position) + SLERP (orientation)
> 
> Submitted an upgrade here: [#1001](https://github.com/huggingface/lerobot/pull/1001) 🚀

Hey, just a note, I think you accidentally Closed the PR instead of leaving it open-

Davegdd, 2025-09-10:
Any news or tutorials on IK for so100?