huggingface/lerobot #679: lerobot/examples /10_use_so100.md  got  "undefined symbol: jpeg12_write_raw_data, version LIBJPEG_8.0"

ylyhlh, 2025-02-04:
### System Info

```Shell
after running the command in instruction
 bash
conda install -y -c conda-forge ffmpeg
pip uninstall -y opencv-python
conda install -y -c conda-forge "opencv>=4.10.0"

Starting to get error, before running that, the teleoperate without camera works just fine,

> libtiff.so.6: undefined symbol: jpeg12_write_raw_data, version LIBJPEG_8.0

Full command and message:

(lerobot_new)  ~/lerobot  ↱ main ±  python lerobot/scripts/control_robot.py \       
  --robot.type=so100 \
  --robot.cameras='{}' \
  --control.type=teleoperate
Traceback (most recent call last):
  File "/home/martinliuhao/lerobot/lerobot/scripts/control_robot.py", line 133, in <module>
    from lerobot.common.robot_devices.control_utils import (
  File "/home/martinliuhao/lerobot/lerobot/common/robot_devices/control_utils.py", line 13, in <module>
    import cv2
  File "/home/martinliuhao/miniconda3/envs/lerobot_new/lib/python3.10/site-packages/cv2/__init__.py", line 181, in <module>
    bootstrap()
  File "/home/martinliuhao/miniconda3/envs/lerobot_new/lib/python3.10/site-packages/cv2/__init__.py", line 153, in bootstrap
    native_module = importlib.import_module("cv2")
  File "/home/martinliuhao/miniconda3/envs/lerobot_new/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: /home/martinliuhao/miniconda3/envs/lerobot_new/lib/python3.10/site-packages/cv2/python-3.10/../../../.././libtiff.so.6: undefined symbol: jpeg12_write_raw_data, version LIBJPEG_8.0
```

### Information

- [x] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

1. running the command in instruction
``` bash
conda install -y -c conda-forge ffmpeg
pip uninstall -y opencv-python
conda install -y -c conda-forge "opencv>=4.10.0"
```
2. teleoperate
``` bash
python lerobot/scripts/control_robot.py \
  --robot.type=so100 \
  --control.type=teleoperate
```

### Expected behavior

The teleoperate should start without error?

shantanuparab-tr, 2025-02-04:
Try this, solved the problem for me.
`conda install -c conda-forge jpeg libtiff
`

ylyhlh, 2025-02-06:
Thanks a lot! It works!
Wondering if the tutorial instruction should be updated?

Cadene, 2025-03-19:
Actually if you run `conda install -c conda-forge jpeg libtiff` you might be blocked later on, since it removes ffmpeg and av1 encoder.

Instead run `pip install torchvision==0.20.1` as a temporary fix.

Thanks @shantanuparab-tr for finding this workaround.

We are on it with @imstevenpmwork to find a fix long term.

closed 2025-03-24 by aliberts

gu021, 2025-04-02:
Hi, now what will be the best solution? After I did " conda install -c conda-forge jpeg libtiff ", my ffmpeg was down-graded from 7.1.1 to 6.1.1. The 6.1.1 version does not enable the "libsvtav1" encoder. This wil introduce problem in recording when doing "episode_000.mp4". 
I checked torchvision version. It is 0.21. Will down grade it to 0.20.1 fix the "LIBJPEG_8" issue in control_robot.py?

shantanuparab-tr, 2025-04-02:
What you should do according to the new update is the following:

1. **Fetch the latest changes** (if you haven't already done that).

2. **Create a new virtual environment using conda**:
   ```bash
   conda create -n lerobot python=3.10 -y
   ```

3. **Install ffmpeg using conda**. This is in accordance with the README, just a few additions to ensures ffmpeg version ≥ 7:
   ```bash
   conda install -c conda-forge 'ffmpeg>=7.0'
   ```

4. **Install lerobot** (change the extras based on the robot you're using; this one is for Aloha):
   ```bash
   pip install --no-binary=av -e ".[dynamixel, intelrealsense]"
   ```

These steps should be enough to avoid any issues with encoders or `jpeg12_write_raw_data`.

gu021, 2025-04-02:
Thanks shantanuparab! I will give it a shot.