huggingface/lerobot #685: Problem with calibration and I'm sure that I've successfully connected to the power supply but it still shows no feedback data for a packet.

Zekai261244292, 2025-02-06:
(lerobot2) jinzekai@jinzekai-Alienware-m15-R6:~/lerobot-main$ python lerobot/scripts/control_robot.py teleoperate \
    --robot-path lerobot/configs/robot/so100.yaml \
    --robot-overrides '~cameras' \
    --display-cameras 0
Connecting main follower arm.
Connecting main leader arm.
Missing calibration file '.cache/calibration/so100/main_follower.json'
Traceback (most recent call last):
  File "/home/jinzekai/lerobot-main/lerobot/scripts/control_robot.py", line 564, in <module>
    teleoperate(robot, **kwargs)
  File "/home/jinzekai/lerobot-joycon/lerobot/common/robot_devices/utils.py", line 28, in wrapper
    raise e
  File "/home/jinzekai/lerobot-joycon/lerobot/common/robot_devices/utils.py", line 24, in wrapper
    return func(robot, *args, **kwargs)
  File "/home/jinzekai/lerobot-main/lerobot/scripts/control_robot.py", line 177, in teleoperate
    control_loop(
  File "/home/jinzekai/lerobot-joycon/lerobot/common/datasets/image_writer.py", line 36, in wrapper
    raise e
  File "/home/jinzekai/lerobot-joycon/lerobot/common/datasets/image_writer.py", line 29, in wrapper
    return func(*args, **kwargs)
  File "/home/jinzekai/lerobot-joycon/lerobot/common/robot_devices/control_utils.py", line 242, in control_loop
    robot.connect()
  File "/home/jinzekai/lerobot-joycon/lerobot/common/robot_devices/robots/manipulator.py", line 331, in connect
    self.activate_calibration()
  File "/home/jinzekai/lerobot-joycon/lerobot/common/robot_devices/robots/manipulator.py", line 406, in activate_calibration
    calibration = load_or_run_calibration_(name, arm, "follower")
  File "/home/jinzekai/lerobot-joycon/lerobot/common/robot_devices/robots/manipulator.py", line 396, in load_or_run_calibration_
    calibration = run_arm_manual_calibration(arm, self.robot_type, name, arm_type)
  File "/home/jinzekai/lerobot-joycon/lerobot/common/robot_devices/robots/feetech_calibration.py", line 423, in run_arm_manual_calibration
    if (arm.read("Torque_Enable") != TorqueMode.DISABLED.value).any():
  File "/home/jinzekai/lerobot-joycon/lerobot/common/robot_devices/motors/feetech.py", line 740, in read
    raise ConnectionError(
ConnectionError: Read failed due to communication error on port /dev/ttyACM1 for group_key Torque_Enable_shoulder_pan_shoulder_lift_elbow_flex_wrist_flex_wrist_roll_gripper: [TxRxResult] There is no status packet!
(lerobot2) jinzekai@jinzekai-Alienware-m15-R6:~/lerobot-main$

closed 2025-02-09 by Zekai261244292

Templarrr, 2025-02-12:
@Zekai261244292 I see you've closed this issue, what did you do to resolve it? I see the very same error

Templarrr, 2025-02-12:
Ok, whoever will face it - for me the problem was the broken power cable between board and first servo - it looked ok but basically one of the cable disattached itself from the connector

vikm2o, 2025-03-15:
thanks @Templarrr I also faced the same issue!

djmango, 2025-04-29:
Ran into this - ensure you plug in the power supply THEN the usb - worked for me!