huggingface/lerobot #687: Annotation number mistake in 1_load_lerobot_dataset.py

hnl1, 2025-02-06:
[1_load_lerobot_dataset.py L132](https://github.com/huggingface/lerobot/blob/638d411cd3acf32c28d8c2120f3c41bda8bb15d4/examples/1_load_lerobot_dataset.py#L132)

```python
    print(f"{batch['observation.state'].shape=}")  # (32, 5, c)
```

I started learning LeRobot today. I noticed that the "5" in the comment here should be "6" because there are 6 timestamps in the observation.state of delta_timestamps earlier.

I'm not sure if this counts as an bug and whether it's worth creating a pull request (PR). However, as a beginner like me, it took me a while to confirm that it was a problem with the comments rather than an issue with my program's output.

hnl1, 2025-02-07:
I found another problem at https://github.com/huggingface/lerobot/blob/638d411cd3acf32c28d8c2120f3c41bda8bb15d4/examples/2_evaluate_pretrained_policy.py#L7C1-L7C27

There is an extra backtick (`) at the end, which causes the command execution to fail.

tc-huang, 2025-02-07:
Hello @hnl1,

I noticed that the issue in `examples/1_load_lerobot_dataset.py` at line 132, which you mentioned, appears to have already been reported in another open pull request (#631).

hnl1, 2025-02-07:
@tc-huang You're right. I just checked PR #631 , and the changes indeed solved the problem of [1_load_lerobot_dataset.py L132](https://github.com/huggingface/lerobot/blob/638d411cd3acf32c28d8c2120f3c41bda8bb15d4/examples/1_load_lerobot_dataset.py#L132)

hnl1, 2025-02-07:
I found another problem at https://github.com/huggingface/lerobot/blob/638d411cd3acf32c28d8c2120f3c41bda8bb15d4/examples/2_evaluate_pretrained_policy.py#L73

The original image already has the channels at the end. The operation here is to move the channels to the front. The annotation is incorrect.

hnl1, 2025-02-07:
I will submit a PR to fix the other issues I've found, and then close this issue.

imstevenpmwork, 2025-04-07:
Thanks! Just merged the fix in https://github.com/huggingface/lerobot/pull/631 and the extra backtick is removed in this one: https://github.com/huggingface/lerobot/pull/948

closed 2025-04-07 by imstevenpmwork