huggingface/lerobot #698: finetune on a local dataset format like aloha

AbdElrahmanMostafaRifaat1432, 2025-02-08:
I want to finetune on a dataset like aloha format and it doesn't contain instruction for each episode but I want to fine tune locally without having repo_id as the following command requires

python lerobot/scripts/train.py
--policy.path=lerobot/pi0
--dataset.repo_id=danaaubakirova/koch_test

Cadene, 2025-02-09:
I am sorry I didnt understand your question. Could you reformulate?

Here is an example to train on a aloha dataset
```bash
python lerobot/scripts/train.py
--policy.path=lerobot/pi0
--dataset.repo_id=lerobot/aloha_sim_transfer_cube
```

AbdElrahmanMostafaRifaat1432, 2025-02-09:
I want to train on a data  on my local without needing to push my dataset to hugging face knowing that I am working on a different dataset other than collected by aloha themselves but it's the same structure

vaishanth-rmrj, 2025-02-21:
I also had the issue of not being able to train the policy using the local dataset without pushing it to the hub. I found a workaround by modifying one of the package files `lerobot/common/datasets/factory.py`.  At line 88 I made the following modifications: 

```
dataset = LeRobotDataset(
      cfg.dataset.repo_id,
      episodes=cfg.dataset.episodes,
      root="<absolute path to your lerobot package>"+str(cfg.dataset.repo_id), # modified line
      delta_timestamps=delta_timestamps,
      image_transforms=image_transforms,
      video_backend=cfg.dataset.video_backend,
      local_files_only=cfg.dataset.local_files_only,
)
```

AbdElrahmanMostafaRifaat1432, 2025-02-21:
@vaishanth-rmrj thanks a lot for your answer

aliberts, 2025-02-25:
Closing this as this should be resolved with #765

closed 2025-02-25 by aliberts