huggingface/lerobot #700: imshow not working

carolineys, 2025-02-09:
When I ran python lerobot/scripts/control_robot.py --robot.type=koch --control.type=teleoperate, the program hangs at (also no image display pop up) the imshow line in ./lerobot/common/robot_devices/control_utils.py

        if display_cameras and not is_headless():
            image_keys = [key for key in observation if "image" in key]
            for key in image_keys:
                cv2.imshow(key, cv2.cvtColor(observation[key].numpy(), cv2.COLOR_RGB2BGR))
            cv2.waitKey(1)

I am using ubuntu 22 and have opencv-python==4.10.0.84 installed.

liyitenga, 2025-02-09:
Have you tried executing this command?
```
pip uninstall -y opencv-python
conda install -y -c conda-forge "opencv>=4.10.0"
```

carolineys, 2025-02-10:
I just figured it out. I originally used a Python virtual environment, and when I reinstalled everything with conda, it worked!

closed 2025-02-10 by carolineys