huggingface/lerobot #705: SO-100 with error:Unknown encoder 'libsvtav1'

HANQAQ, 2025-02-10:
When I use the new version of Lerobot's SO-100 and run the following command:
`python lerobot/scripts/control_robot.py \
  --robot.type=so100 \
  --control.type=record \
  --control.fps=30 \
  --control.single_task="Grasp a lego block and put it in the bin." \
  --control.repo_id=${HF_USER}/so100_test \
  --control.tags='["so100","tutorial"]' \
  --control.warmup_time_s=5 \
  --control.episode_time_s=30 \
  --control.reset_time_s=30 \
  --control.num_episodes=2 \
  --control.push_to_hub=true`
the following errors occur:
`[vost#0:0 @ 0x13207240] Unknown encoder 'libsvtav1'
[vost#0:0 @ 0x13207240] Error selecting an encoder
Error opening output file /home/han/.cache/huggingface/lerobot/lyhhan/so100_test/videos/chunk-000/observation.images.laptop/episode_000000.mp4.
Error opening output files: Encoder not found`

but, when i use the old version, this error does not occur.

RuiboLiu77, 2025-02-10:
How I solved this problem was by following the instructions to create the virtual environment and installed required pkg.
After that, I ran 
 conda install -y -c conda-forge ffmpeg 
and skipped the next two commands (pip uninstall -y opencv-python
conda install -y -c conda-forge "opencv>=4.10.0").

HANQAQ, 2025-02-10:
> How I solved this problem was by following the instructions to create the virtual environment and installed required pkg. After that, I ran conda install -y -c conda-forge ffmpeg and skipped the next two commands (pip uninstall -y opencv-python conda install -y -c conda-forge "opencv>=4.10.0").

It seems that this method doesn't work.

tlpss, 2025-02-10:
I also stumbled upon this issue.

You can change the video codec to e.g. 'x265' if you cannot install a more recent version of ffmpeg (>5.1 is required for AV1 as stated in [the docs](https://trac.ffmpeg.org/wiki/Encode/AV1)). 

The default apt package repository does not include recent ffmpeg releases for ubuntu 20/22 (cf [here](https://askubuntu.com/questions/1497843/how-to-install-latest-version-of-ffmpeg-on-ubuntu)). You can install from source or consider another package repository.

HANQAQ, 2025-02-10:
> I also stumbled upon this issue.
>
> You can change the video codec to e.g. 'x265' if you cannot install a more recent version of ffmpeg (>5.1 is required for AV1 as stated in [the [...]

i try libsvtav1, but it will get stuck during video encoding, resulting in the failure of data collection.

carolineys, 2025-02-10:
I am getting the same error, did you solve it?

HANQAQ, 2025-02-11:
> > > I also stumbled upon this issue.
> > > You can change the video codec to e.g. 'x265' if you cannot install a more recent version of ffmpeg (>5.1 is required for AV1 as stated in [the docs](https://trac.ffmpeg.org/wiki/Encode/AV1)).
> > > The default apt package repository does not include recent ffmpeg releases for ubuntu 20/22 (cf [here](https://askubuntu.com/questions/1497843/how-to-install-latest-version-of-ffmpeg-on-ubuntu)). You can install from source or consider another package repository.
> > 
> > 
> > i try libsvtav1, but it will get stuck during video encoding, resulting in the failure of data collection.
> 
> Have you verified your ffmpeg version? If you cannot use AV1 and cannot upgrade ffmpeg, you can change the video codec [here](https://github.com/huggingface/lerobot/blob/334deb985dc8f4822fd26712e76bb5ab47c5d01d/lerobot/common/datasets/video_utils.py#L134) to `vcodec: str = "libx264"` to use the X.264 codec, which should be available in ffmpeg (check with `ffmpeg -codecs | grep 264`)

I re-git cloned "lerobot" and recreated the conda environment. Although I found that I didn't have `libx264`, I discovered that I had `libopenh264`. Then, after trying `libopenh264`, it worked. Thank you very much. :)

yangcyself, 2025-04-30:
`libx264` is not passing the test of `vcodec` [here](https://github.com/huggingface/lerobot/blob/6d723c45a993d1e030dfda20d48a893a74a4fe11/lerobot/common/datasets/video_utils.py#L259) in current main branch 
I solved the problem by `pip install "av>=14"`

anarnuri, 2025-08-21:
@yangcyself thank you, it solved my issue too, other solutions unfortunately did not work.