huggingface/lerobot #713: AttributeError: 'list' object has no attribute 'to'

Tramac, 2025-02-11:
When run the code `python3 ./examples/3_train_policy.py`, the following error is reported:
```
Traceback (most recent call last):
  File "./lerobot/examples/3_train_policy.py", line 121, in <module>
    main()
  File "./lerobot/examples/3_train_policy.py", line 102, in main
    batch = {k: v.to(device, non_blocking=True) for k, v in batch.items()}
  File "/data/code/lerobot/examples/3_train_policy.py", line 102, in <dictcomp>
    batch = {k: v.to(device, non_blocking=True) for k, v in batch.items()}
AttributeError: 'list' object has no attribute 'to'
```

Because the data type of `batch['task']` is `list`, such as:
```
['Push the T-shaped block onto the T-shaped target.',                   
 'Push the T-shaped block onto the T-shaped target.',                                                    
 'Push the T-shaped block onto the T-shaped target.',                   
 'Push the T-shaped block onto the T-shaped target.',                                                    
 'Push the T-shaped block onto the T-shaped target.',                                  
 'Push the T-shaped block onto the T-shaped target.',                                                    
 'Push the T-shaped block onto the T-shaped target.',
...
]
```

Maybe the [code](https://github.com/huggingface/lerobot/blob/main/examples/3_train_policy.py#L88) should be:
```
batch = {k: (v.to(device, non_blocking=True) if isinstance(v, torch.Tensor) else v) for k, v in batch.items()}
```

brysonjones, 2025-02-16:
Can confirm I also hit this error and the suggested fix works.

imstevenpmwork, 2025-04-07:
This was solved in: https://github.com/huggingface/lerobot/pull/799

closed 2025-04-07 by imstevenpmwork