huggingface/lerobot #717: gstreamer error

carolineys, 2025-02-12:
When I connected to one camera, everything works fine. But when I try to connect to two cameras, I got the following errors:
[ WARN:0@0.048] global cap_gstreamer.cpp:1173 isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
[ WARN:0@0.169] global cap_gstreamer.cpp:1173 isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
Error: Failed to capture image from camera 2.

vaishanth-rmrj, 2025-02-21:
Have you checked the camera port id configuration for your robot?  You can get the camera port id's using this script: `lerobot/common/robot_devices/cameras/opencv.py`.

fireman5379, 2025-03-23:
> Have you checked the camera port id configuration for your robot? You can get the camera port id's using this script: `lerobot/common/robot_devices/cameras/opencv.py`.

the error just happend when using this script. BUT I can run the code before, TODAY the error happend , SO weird!

fireman5379, 2025-03-23:
Just unplug a usb cable of camera and plug it to another pc's usb port.Because the bandwith of a usb 2.0 port is not enough.
YOU CAN SEE https://blog.csdn.net/waynebuaa/article/details/41786281
IT WORKED FOR ME!

CarolinePascal, 2025-04-14:
Hi @carolineys !

Were you able to reproduce this issue ? If so, have you tried switching from a 2.0 USB connection to a 3.0 USB connection as mentioned by @fireman5379 ?

Best,

Caroline.