huggingface/lerobot #723: Weird when loading dataset with delta_timestamps param specified

feixyz10, 2025-02-13:
### System Info

```Shell
- `lerobot` version: 0.1.0
- Platform: macOS-14.6.1-arm64-arm-64bit
- Python version: 3.12.8
- Huggingface_hub version: 0.28.1
- Dataset version: 3.2.0
- Numpy version: 2.1.3
- PyTorch version (GPU?): 2.6.0 (False)
- Cuda version: N/A
- Using GPU in script?: <fill in>
```

### Information

- [ ] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

> 1_load_lerobot_dataset.py

```python
repo_id = "lerobot/aloha_mobile_cabinet"
delta_timestamps = {
    camera_key: [-1, -0.5, -0.20, 0],
    "observation.state": [-1.5, -1, -0.5, -0.20, -0.10, 0],
    "action": [t / dataset.fps for t in range(64)],
}
dataset = LeRobotDataset(repo_id, episodes=[0, 10, 11, 23], delta_timestamps=delta_timestamps)
camera_key = dataset.meta.camera_keys[0]

print([dataset[i][camera_key+"_is_pad"] for i in range(10)])
print([torch.allclose(dataset[i][camera_key][3], dataset[i][camera_key][2]) for i in range(10)])

````

**current outputs:**

[tensor([ True,  True,  True, False]), tensor([ True,  True,  True, False]), tensor([ True,  True,  True, False]), tensor([ True,  True,  True, False]), tensor([ True,  True,  True, False]), tensor([ True,  True,  True, False]), tensor([ True,  True,  True, False]), tensor([ True,  True,  True, False]), tensor([ True,  True,  True, False]), tensor([ True,  True,  True, False])]

[True, True, False, False, False, False, False, False, False, False]

### Expected behavior

Output for the line `print([dataset[i][camera_key+"_is_pad"] for i in range(10)])` seems being correct, but the output for `torch.allclose` not. If I understand correctly, the output for `torch.allclose` should be all 'True'.

aliberts, 2025-03-02:
This dataset's fps is 50. 
This means that for the first 10 indices, without any delta_timestamp you'd get frames between `0`s and `9*1/50`=`0.18`s

With `delta_timestamps = {camera_key: [-1, -0.5, -0.20, 0]}`, `dataset[i][camera_key][2]` is outside that range for `i in range(10)` and so they are padded while `dataset[i][camera_key][3]` is simply the `i`th frame, so it's not padded.

Now normally we would expect to see the second element of the list
```python
[torch.allclose(dataset[i][camera_key][3], dataset[i][camera_key][2]) for i in range(10)]
```
to be `False` because it's comparing the first frame with the second one, but it just so happens that these 2 frames in this case are close enough so that the check passes (you can [download the video](https://huggingface.co/datasets/lerobot/aloha_mobile_cabinet/resolve/main/videos/chunk-000/observation.images.cam_high/episode_000000.mp4?download=true) and visually inspect it for yourself). This is because when frame are decoded, they are first loaded as `uint8` by `torchvision.io.VideoReader` so the differences are rounded at that point. They are only then converted to float32, which explains why the `torch.allclose` is True in that case even though the 2 frames are not exactly the same.

closed 2025-03-02 by aliberts

feixyz10, 2025-03-02:
@aliberts Thanks a lot for your reply! 

I understood why first 3 images in a mini-batch are needed to pad, its reason can also explain why `dataset[i][camera_key+"_is_pad` yields `tensor([ True,  True,  True, False])` for $i \in [0, 1, ..., 9]$. But I am confused that if they (first 3 images) are all padded by copying the 4-th image, why `torch.allclose(dataset[i][camera_key][3], dataset[i][camera_key][2])` is not evaluated to `True` for all $i \in [0, 1, ..., 9]$?  BTW, what padding strategy ([mode](https://numpy.org/doc/stable/reference/generated/numpy.pad.html)) is used in LeRobotDataset,