huggingface/lerobot #727: Pass desired vcodec in LeRobotDataset.encode_episode_videos

fabio-amadio, 2025-02-13:
First of all thanks a lot for the amazing job done with lerobot!

Trying out the example script [**pusth_zarr.py**](https://github.com/huggingface/lerobot/blob/main/examples/port_datasets/pusht_zarr.py) I am getting this error from ffmpeg: *Unknown encoder 'libsvtav1'* (when in "video" mode).

To avoid this issue I selected another encoder by adding the argument `vcodec = "libx264"` in the  [`encode_vide_frames` call inside `encode_episode_videos`](https://github.com/huggingface/lerobot/blob/e71095960f3b2970e0915e6a5be7bd006b266173/lerobot/common/datasets/lerobot_dataset.py#L883) . 

What's the proper way of passing the vcodec argument so I can avoid hard-coding the desired value inside **lerobot_dataset.py**?

Tavish9, 2025-02-15:
Hi, @fabio-amadio, you use compile ffmpeg to use "libsvtav1" encoder, see [this].(https://trac.ffmpeg.org/wiki/CompilationGuide)

fabio-amadio, 2025-02-15:
Hi @Tavish9, thanks! I am now able to use "libstav1".

closed 2025-03-02 by aliberts