huggingface/lerobot #745: Error occured during evaluate phase

FanZhang91, 2025-02-18:
### System Info

```Shell
ubuntu==20.04
lerobot==0.1.0
gym-aloha==0.1.1
gym-pusht=0.1.5
gym-xarm==0.1.1
gymnasium==0.29.1
```

### Information

- [ ] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

**I use the following script to run the pi0 training program, but encountered the following error。

script:  python train.py --dataset.repo_id=lerobot/aloha_sim_insertion_human --env.type=aloha --env.task=AlohaInsertion-v0 --policy.type=pi0**

---------------------------------------------------- error info ---------------------------------------------------------------
WARNING 2025-02-18 20:15:03 gs/train.py:129 No device specified, trying to infer device automatically
INFO 2025-02-18 20:15:03 ils/utils.py:46 Cuda backend detected, using cuda.
INFO 2025-02-18 20:15:03 ts/train.py:192 {'batch_size': 8,
 'dataset': {'episodes': None,
             'image_transforms': {'enable': False,
                                  'max_num_transforms': 3,
                                  'random_order': False,
                                  'tfs': {'brightness': {'kwargs': {'brightness': (0.8,
                                                                                   1.2)},
                                                         'type': 'ColorJitter',
                                                         'weight': 1.0},
                                          'contrast': {'kwargs': {'contrast': (0.8,
                                                                               1.2)},
                                                       'type': 'ColorJitter',
                                                       'weight': 1.0},
                                          'hue': {'kwargs': {'hue': (-0.05,
                                                                     0.05)},
                                                  'type': 'ColorJitter',
                                                  'weight': 1.0},
                                          'saturation': {'kwargs': {'saturation': (0.5,
                                                                                   1.5)},
                                                         'type': 'ColorJitter',
                                                         'weight': 1.0},
                                          'sharpness': {'kwargs': {'sharpness': (0.5,
                                                                                 1.5)},
                                                        'type': 'SharpnessJitter',
                                                        'weight': 1.0}}},
             'local_files_only': False,
             'repo_id': 'lerobot/aloha_sim_insertion_human',
             'use_imagenet_stats': True,
             'video_backend': 'pyav'},
 'device': 'cuda',
 'env': {'episode_length': 400,
         'features': {'action': {'shape': (14,),
                                 'type': <FeatureType.ACTION: 'ACTION'>},
                      'agent_pos': {'shape': (14,),
                                    'type': <FeatureType.STATE: 'STATE'>},
                      'pixels/top': {'shape': (480, 640, 3),
                                     'type': <FeatureType.VISUAL: 'VISUAL'>}},
         'features_map': {'action': 'action',
                          'agent_pos': 'observation.state',
                          'pixels/top': 'observation.images.top',
                          'top': 'observation.image.top'},
         'fps': 50,
         'obs_type': 'pixels_agent_pos',
         'render_mode': 'rgb_array',
         'task': 'AlohaInsertion-v0'},
[... 85 lines cut ...]
INFO 2025-02-18 20:15:35 ts/train.py:143 step:0 smpl:8 ep:0 epch:0.00 loss:1.837 grdn:12.157 lr:5.0e-08 updt_s:0.840 data_s:0.350
INFO 2025-02-18 20:15:35 ts/train.py:252 Eval policy at step 1
Stepping through eval batches:   0%|          | 0/1 [00:00<?, ?it/s]
Stepping through eval batches:   0%|          | 0/1 [00:01<?, ?it/s]
Traceback (most recent call last):
  File "/home/xxx/deep_learning/lerobot/lerobot/scripts/train.py", line 254, in evaluate_and_checkpoint_if_needed
    eval_info = eval_policy(
  File "/home/xxx/deep_learning/lerobot/lerobot/scripts/eval.py", line 295, in eval_policy
    rollout_data = rollout(
  File "/home/xxx/deep_learning/lerobot/lerobot/scripts/eval.py", line 160, in rollout
    action = policy.select_action(observation)
  File "/home/xxx/miniforge3/envs/lerobot/lib/python3.10/site-packages/torch/utils/_contextlib.py", line 116, in decorate_context
    return func(*args, **kwargs)
  File "/home/xxx/deep_learning/lerobot/lerobot/common/policies/pi0/modeling_pi0.py", line 285, in select_action
    lang_tokens, lang_masks = self.prepare_language(batch)
  File "/home/xxx/deep_learning/lerobot/lerobot/common/policies/pi0/modeling_pi0.py", line 386, in prepare_language
    tasks = batch["task"]
KeyError: 'task'
------------------------------------------------------ error info -------------------------------------------------------------------------

**In the evaluation phase, “observation” information comes from simulation environment. How to fix this bug?**

### Expected behavior

null

liyitenga, 2025-02-26:
I have the same problem

zlsjsj, 2025-03-07:
I have the same problem

frk2, 2025-03-23:
And same problem!