huggingface/lerobot #746: How should I run the model on my own datasets in different envs which is not clearly mentioned in the README?

shi-akihi, 2025-02-18:
I want to run the diffusion model on my own real world arms datasets, which are different from the example env and input format in observation and action dims.

I've seem some yaml files to store these parameters in earlier version of the repo, but I can't find it in the newest version of the repo. So should I write this params myself in some yaml-like or json-like files or there are some new ways to solve these problems. 

This is my first issue in github, so the format may be informal, but I'm really eager for the answers. 
Thank you for your answers!!!

vaishanth-rmrj, 2025-02-20:
The LeRobot package is tightly integrated to use LeRobot Dataset. You can try converting your dataset to LeRobot dataset format. Example script: `lerobot/exaples/port_datasets/pusht_zarr.py`. Also make sure your robot is supported by the package.

shi-akihi, 2025-03-03:
Thank you. I knew that the dataset should be set in LeRobot Dataset format. Well, the real problem that bothering me is that I don't find any config file to modify that exact training parameters I want to set. 
For example, if I have a different action space dim from the existing choice, how should I modify it? I think LeRobot don't load these parameters from the meta data of LeRobot format dataset.

closed 2025-10-19 by github-actions[bot]