huggingface/lerobot #761: What should the .parquet and .json files in LeRobotDataset directory look like?

pqrsqwewrty, 2025-02-22:
I'm reading through the documentation for LeRobotDataset, and it says that the root dataset directory structure should look like this:
```
A typical LeRobotDataset looks like this from its root path:
.
├── data
│   ├── chunk-000
│   │   ├── episode_000000.parquet
│   │   ├── episode_000001.parquet
│   │   ├── episode_000002.parquet
│   │   └── ...
│   ├── chunk-001
│   │   ├── episode_001000.parquet
│   │   ├── episode_001001.parquet
│   │   ├── episode_001002.parquet
│   │   └── ...
│   └── ...
├── meta
│   ├── episodes.jsonl
│   ├── info.json
│   ├── stats.json
│   └── tasks.jsonl
└── videos
	├── chunk-000
	│   ├── observation.images.laptop
	│   │   ├── episode_000000.mp4
	│   │   ├── episode_000001.mp4
	│   │   ├── episode_000002.mp4
	│   │   └── ...
	│   ├── observation.images.phone
	│   │   ├── episode_000000.mp4
	│   │   ├── episode_000001.mp4
	│   │   ├── episode_000002.mp4
	│   │   └── ...
	├── chunk-001
	└── ...
```

Questions:
1) What should the structure of episodes.json, info.json, stats.json, tasks.json look like?
2) What should the structure of each episode_xxxxxx.parquet file look like?
3) What does chunking the data do? Is there a limit to how many episodes per chunk?

Tavish9, 2025-02-22:
Hi, @pqrsqwewrty, why not try [openx2lerobot](https://github.com/Tavish9/openx2lerobot)? It's very simple and you can just change the source (like rlds, hdf5, etc.) and everything is done. You don't have to worry about field like `next.reward`. Instead, just care about `observation` and `action`.

closed 2025-02-25 by pqrsqwewrty

arpitg1304, 2026-01-30:
Single command solution that you can give a try

https://github.com/arpitg1304/forge/tree/main

forge convert /path/to/rlds ./output --format lerobot-v3
forge convert hf://arpitg1304/stack_lego ./stack_lego_rlds --format rlds --workers 4 --visualize
forge convert hf://lerobot/pusht ./pusht_robodm --format robodm