huggingface/lerobot #773: How to overrite the code to collect action datas from others robot？

tjh-flash, 2025-02-26:
Hey，I have got a problem when i try to overwrite the code of lerobot to collect action datas from my own robot. Here‘s the detail. My robot is a single six joint robot arm, so i make a new RobotConfig, which only contains the info of the camera. And then I overwrite the fuction 'teleop_step' in file manipulator.py. I also set a default value of the robot pos to test at first.  When i start to record,  the datad of observation and action  are fine, but when it comes to call the function 'save_eposide',  error comes up, which i show below. I reall want to know what else should i suppose to do to make it work, thanks.

[image]
[image]
[image]

tjh-flash, 2025-02-26:
Here is the error info.
[image]

imstevenpmwork, 2025-04-03:
Hello @tjh-flash , this seems like a bug in your custom logic

closed 2025-10-16 by github-actions[bot]