huggingface/lerobot #805: Program Hangs at "Activating torque on main follower arm." During Teleoperation SO 100

winster-bai, 2025-03-04:
### System Info

```Shell
Dear LeRobot Development Team,

I am experiencing an issue with the SO100 robotic arm during teleoperation. When executing the command python lerobot/scripts/control_robot.py --robot.type=so100 --control.type=teleoperate, the system hangs at the "Activating torque on main follower arm." stage.

System Information:

Operating System: Ubuntu 22.04
CUDA Version: 12.3
Python Version: 3.10
GPU: NVIDIA RTX 4090
Log Details are as followed

Troubleshooting Steps Taken:

Checked Power Connections: Verified that the main follower arm's power supply is properly connected and functioning.
Motor Configuration: Ensured that all motors are correctly configured with unique IDs as per the LeRobot guidelines.
Set Device Permissions: Applied sudo chmod 666 /dev/ttyACM0 to grant appropriate permissions to the device.
Verified Dependencies: Confirmed that all necessary dependencies are installed and up-to-date.

Additional Notes:

I have ensured that the servo motors are correctly oriented and calibrated. However, the issue persists, and the system consistently hangs at the same point. I have also consulted the official LeRobot documentation and followed the setup instructions meticulously.

I would appreciate any assistance or guidance you can provide to resolve this issue.

Thank you for your support.

Best regards,
Winster
```

### Information

- [x] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

'''
/.local/lib/python3.10/site-packages/torch/utils/_pytree.py:185: FutureWarning: optree is installed but the version is too old to support PyTorch Dynamo in C++ pytree. C++ pytree support is disabled. Please consider upgrading optree using `python3 -m pip install --upgrade 'optree>=0.13.0'`.
  warnings.warn(
INFO 2025-03-04 16:57:42 ol_robot.py:358 {'control': {'display_cameras': True, 'fps': None, 'teleop_time_s': None},
 'robot': {'calibration_dir': '.cache/calibration/so100',
           'cameras': {'laptop': {'camera_index': 0,
                                  'channels': 3,
                                  'color_mode': 'rgb',
                                  'fps': 30,
                                  'height': 480,
                                  'mock': False,
                                  'rotation': None,
                                  'width': 640},
                       'phone': {'camera_index': 2,
                                 'channels': 3,
                                 'color_mode': 'rgb',
                                 'fps': 30,
                                 'height': 480,
                                 'mock': False,
                                 'rotation': None,
                                 'width': 640}},
           'follower_arms': {'main': {'mock': False,
                                      'motors': {'elbow_flex': [3, 'sts3215'],
                                                 'gripper': [6, 'sts3215'],
                                                 'shoulder_lift': [2,
                                                                   'sts3215'],
                                                 'shoulder_pan': [1, 'sts3215'],
                                                 'wrist_flex': [4, 'sts3215'],
                                                 'wrist_roll': [5, 'sts3215']},
                                      'port': '/dev/ttyACM0'}},
           'gripper_open_degree': None,
           'leader_arms': {'main': {'mock': False,
                                    'motors': {'elbow_flex': [3, 'sts3215'],
                                               'gripper': [6, 'sts3215'],
                                               'shoulder_lift': [2, 'sts3215'],
                                               'shoulder_pan': [1, 'sts3215'],
                                               'wrist_flex': [4, 'sts3215'],
                                               'wrist_roll': [5, 'sts3215']},
                                    'port': '/dev/ttyACM1'}},
           'max_relative_target': None,
           'mock': False}}
Connecting main follower arm.
Connecting main leader arm.
Activating torque on main follower arm.
'''

### Expected behavior

The system should proceed past the "Activating torque on main follower arm." stage without freezing, allowing for successful teleoperation.

winster-bai, 2025-03-06:
Anyone？

trltzy, 2025-03-09:
Same Problem. Any one solved？

closed 2025-03-12 by winster-bai

ZorAttC, 2025-03-17:
I encountered the same problem and I don't know why.

kazuki-shin, 2025-03-20:
Reinstalling conda env + install libtiff lib worked for me

Yan1205, 2025-03-30:
I'm facing the same issue, and I've identified the root cause: **we're mixing pip and conda installations.**

In most cases, both work fine when used separately, but mixing them will drive you crazy with version conflicts! Conda has strict dependency checking - it ensures all conda-installed packages work together, but it only checks packages installed through conda. For example, when installing a new package that depends on numpy, if you already have numpy (say 1.19.2) installed via pip, conda won't recognize it and may install another version (e.g., 1.18.5). Now you have two numpy versions potentially conflicting.
You might see different numpy versions in pip list and conda list, making it unclear which version your program actually uses. Uninstalling may not clean everything properly - you might remove one but the other remains, or end up with empty shell packages. The quick solution? Manually delete numpy from anaconda's lib/site-packages/numpy!
Pip does check both pip-installed and conda-installed packages during installation, but it simply installs what's requested without ensuring all packages work together - they might install but not function properly.
The golden rule:
    In a new environment, if you need both conda and pip packages, always install conda packages first, then pip.
    If possible, stick to just one package manager throughout.

The true solution is to **install opencv,numpy exclusively through conda.**

ZorAttC, 2025-06-02:
> For me, what fixed it was:
>  --control.display_cameras=false

Thx!I solve this problem by buying a longer cable for the SO100 to avoid loosen.