huggingface/lerobot #808: How to acquire the End-Effector（eef） pose?

oym1994, 2025-03-04:
Hi, thanks for your great job!

    How can we acquire the eef pose and control the eef pose instead of only the joints states?

Thanks for your attention and hope for your kind response!

mohitydv09, 2025-03-09:
Which robot are you using?

oym1994, 2025-03-11:
> Which robot are you using?

SO100

oym1994, 2025-03-12:
> I'm trying to use the end-effector pose (SE3) as the action space, and then solve for joint angles using IK, but I've noticed that due to the robot arm's degree of freedom limitations, many poses have no IK solution. This causes the trained model to easily generate unsolvable poses, resulting in the robot arm getting stuck and not moving. Does anyone have a better solution for this issue?

hello, which ik package do you use?

Buzz-Beater, 2025-03-31:
Hi, any recommendation on which IK solver to be used on the SO-100 arm? I'm trying the same thing to use end-effector position as the action space