huggingface/lerobot #817: SO 100 Arm assembly instruction conflict?

liuhuanjim013, 2025-03-05:
Step 22 of the assembly guide shows a picture of wrist that is flipped comparing to the drawing and front page photo. Are both right? If not, which one is correct?

[Latest instruction](https://github.com/huggingface/lerobot/blob/main/examples/10_use_so100.md#wrist-assembly):
[image]

[Assembly video](https://github.com/huggingface/lerobot/blob/main/examples/10_use_so100.md#additional-guidance):
[image]

[Project home page](https://github.com/huggingface/lerobot/tree/main?tab=readme-ov-file#------------build-your-own-so-100-robot):
[image]

Pierre-LouisBJT, 2025-03-13:
You can check this video of a full assembly: [https://youtu.be/PG_EJAeNRfM](https://youtu.be/PG_EJAeNRfM)
It should answer your question.

brainwavecoder9, 2025-05-19:
**Short version: Both are correct.**

The photo from the instructions for Step 22 of the assembly guide is reversed for how I have assembled my arm. So I am going to say my way is right :)

In reality, it does not matter. The orientation of the part does not change anything about the physics of the robot; the robot won't care which way the servo is oriented / which way the plastic faces. It may be "wrong" but it doesn't matter because calibration will make it "right". 

The physical location of the important parts of the servo won't change no matter how that part is oriented.

closed 2025-11-30 by github-actions[bot]