huggingface/lerobot #826: Should the pi0 pytorch model on Huggingface load model.safetensors or the other three satetensors?

chopinxxxx, 2025-03-06:
https://huggingface.co/lerobot/pi0/tree/main

What is the difference between `model.safetensors` and the other three satetensors (`model-00001-of-0000*.safetensors`)? The pi0 model `from_pretrained()` method will load `model.safetensor`s by default instead of `model-00001-of-0000*.safetensors`.

ximiluuuu, 2025-03-26:
hi, i meet the same question, do you deal it?

chopinxxxx, 2025-03-27:
> hi, i meet the same question, do you deal it?

no, i still use the model.safetensors file for loading pretrained weights.

pkooij, 2025-10-08:
Hi we update our pi0 implementation and released pi05 support in this PR: https://github.com/huggingface/lerobot/pull/1910 🚀 . If you still have this issue please reopen this thread, but I am closing it for now.