huggingface/lerobot #836: cli overrides failed when use a pretrained model with `--policy.path`

DelinQu, 2025-03-08:
### System Info

```Shell
Lerobot Version: 0.1.0
OS: Ubuntu 22.04.5 LTS x86_64
Python Version: Python 3.10.16
Numpy Version: 1.26.4
Torch Version: 2.6.0
Lerobot Config: 
Draccus Version: 0.10.0
config.json:

{
    "dataset": {
        "repo_id": "libero_spatial_no_noops_image",
        "episodes": null,
        "local_files_only": true
    },
    "policy": {
        "type": "pi0",
        "optimizer_lr": 0.00025
    }
}
```

### Information

- [x] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

Hi, lerobot team! I'm fine-tuning a pre-trained policy pi0 using the following cmd:
```
python lerobot/scripts/train.py \
    --policy.path=../pretrained/pi0 \
    --config_path=config.json \
    --dataset.repo_id=libero_spatial_no_noops_1.0.0_lerobot \
    --wandb.enable=false \
    --policy.optimizer_lr=5e-5
```
The configuration system fails to recognize the parameter `optimizer_lr` when I start training. However, the programs launch successfully if I remove `--policy.path=../pretrained/pi0`. The error information are follows:
```bash
WARNING 2025-03-08 23:05:11 igs/train.py:65 No device specified, trying to infer device automatically
INFO 2025-03-08 23:05:12 ils/utils.py:42 Cuda backend detected, using cuda.
usage: train.py [-h] [--config_path str] [--n_obs_steps str] [--normalization_mapping str] [--input_features str] [--output_features str]
train.py: error: unrecognized arguments: --optimizer_lr=5e-5
```

### Expected behavior

The configuration system is expected to recognize the inner parameter in the policy configuration class, which is more flexible to fine-tuning a policy with different hyperparameters.

The-Michael-Chen, 2025-03-10:
i got the same issue

imstevenpmwork, 2025-03-10:
Thanks for spotting this out! This is a current limitation of the framework we use for the CLI parsing and Configuration management (Draccus). We're considering some work arounds but for now the best way forward is somewhat unclear. I would recommend you to, at least for now, when you use `policiy.path` you try to add all your configurations to the file you are loading such that you don't need to override parameters via the CLI. I ill update this issue when we get an update related to this 😄

brainwavecoder9, 2025-05-19:
Thanks for the update @imstevenpmwork. 

Any insight into the state of discussions? I'm assuming the long term solution is to update draccus to handle overrides, but the team is unsure if they want to consider a lower effort solution such as updating train.py to accept overrides 

I think the issue is that a user may never realize cli overrides don't work. I would consider that to be a silent failure which is quite frustrating to discover after iterative updates to training runs don't produce the expected results. 

How about something in the interim? If you're open to accepting the PR I'd be happy to code up a solution which adds a validation step in the train.py script that:
- Checks for conflicting CLI arguments
- Throws an error message when conflicts are detected
- Guides users to the use of config files for supplying additional CLI params (or the alternative, dropping --policy-path and supplying all necessary params through the CLI)
-

closed 2025-11-30 by github-actions[bot]