huggingface/lerobot #846: How to convert my own dataset to LerobotDataset format?

yilin404, 2025-03-11:
Hi, I am new to Lerobot and have a dataset in my own format. I would like to convert it to the LerobotDataset format.

I referred to lerobot/scripts/push_dataset_to_hub.py, but it seems to be deprecated. Could you provide guidance or an updated method for converting custom datasets?

Thanks in advance!

Tavish9, 2025-03-12:
Hi, I wrote a script for converting OpenX(rlds) format to LeRobot format, please check [openx2lerobot](https://github.com/Tavish9/openx2lerobot). 

If your own dataset is not in rlds format, you can still update the `for` loop for getting your dataset item. Let me know if you need help!

RuiboLiu77, 2025-03-20:
you could check the script from openpi: https://github.com/Physical-Intelligence/openpi/blob/main/examples/aloha_real/convert_aloha_data_to_lerobot.py
you can modify the script from aloha data structure to your owns.

imstevenpmwork, 2025-03-21:
This will depend on the specific format of your dataset. May I ask which format are you using?

closed 2025-03-24 by yilin404

zhangziwenHIT, 2025-04-11:
Can someone share the script for converting Robomind to Lerobot? I'd appreciate it!

Tavish9, 2025-04-11:
> Can someone share the script for converting Robomind to Lerobot? I'd appreciate it!

working on it, also AgiBotWorld😇

zhangziwenHIT, 2025-04-14:
> > Can someone share the script for converting Robomind to Lerobot? I'd appreciate it!
> 
> working on it, also AgiBotWorld😇
> 
> would be released on [any4lerobot](https://github.com/Tavish9/any4lerobot)

That would be great! If possible please post the script first so I can convert it locally, thanks!