huggingface/lerobot #850: draccus.utils.DecodingError due to  The fields `device`, `use_amp` are not valid for ACTConfig in PreTrainedConfig

binliu-base, 2025-03-12:
### System Info

```Shell
- `lerobot` version: 0.1.0
- Platform: Linux-6.8.0-52-generic-x86_64-with-glibc2.35
- Python version: 3.10.13
- Huggingface_hub version: 0.29.1
- Dataset version: 3.3.2
- Numpy version: 1.26.4
- PyTorch version (GPU?): 2.6.0+cu124 (False)
- Cuda version: 12040
- Using GPU in script?: no
```

### Information

- [ ] One of the scripts in the examples/ folder of LeRobot
- [x] My own task or dataset (give details below)

### Reproduction

When I used the following command to evaluate a trained model, I ran into a `draccus.utils.DecodingError`:

Evaluation command:
```bash
python lerobot/scripts/control_robot.py --robot.type=so100 --control.type=record --control.fps=30 --control.single_task="TEST" --control.repo_id=luke250305/eval_play_dice_250311.1 --control.tags='["tutorial"]' --control.warmup_time_s=5 --control.episode_time_s=20 --control.reset_time_s=10 --control.num_episodes=10 --control.push_to_hub=false --control.policy.path=/home/binliu/lerobot.v2/outputs/modles/act_play_dice_250311.1/checkpoints/last/pretrained_model
```

Error:
```plaintext
Traceback (most recent call last):
  File "/home/binliu/lerobot.v2/lerobot/scripts/control_robot.py", line 382, in <module>
    control_robot()
  File "/home/binliu/lerobot/lerobot/configs/parser.py", line 119, in wrapper_inner
    cfg = draccus.parse(config_class=argtype, config_path=config_path, args=cli_args)
  File "/home/binliu/miniconda3/envs/lerobot/lib/python3.10/site-packages/draccus/argparsing.py", line 211, in parse
    return parser.parse_args(args)
  File "/home/binliu/miniconda3/envs/lerobot/lib/python3.10/site-packages/draccus/argparsing.py", line 102, in parse_args
    args, _ = self.parse_known_args(args, namespace, is_parse_args=True)
  File "/home/binliu/miniconda3/envs/lerobot/lib/python3.10/site-packages/draccus/argparsing.py", line 136, in parse_known_args
    parsed_t = self._postprocessing(parsed_args)
  File "/home/binliu/miniconda3/envs/lerobot/lib/python3.10/site-packages/draccus/argparsing.py", line 180, in _postprocessing
    cfg = decoding.decode(self.config_class, deflat_d)
  File "/home/binliu/miniconda3/envs/lerobot/lib/python3.10/site-packages/draccus/parsers/registry_utils.py", line 78, in wrapper
    return base_func(*args, **kw)
  File "/home/binliu/miniconda3/envs/lerobot/lib/python3.10/site-packages/draccus/parsers/decoding.py", line 48, in decode
    return get_decoding_fn(cls)(raw_value, ())  # type: ignore
  File "/home/binliu/miniconda3/envs/lerobot/lib/python3.10/site-packages/draccus/parsers/decoding.py", line 135, in decode_dataclass
    raise e
  File "/home/binliu/miniconda3/envs/lerobot/lib/python3.10/site-packages/draccus/parsers/decoding.py", line 133, in decode_dataclass
    field_value = get_decoding_fn(field_type)(raw_value, (*path, name))
  File "/home/binliu/miniconda3/envs/lerobot/lib/python3.10/site-packages/draccus/parsers/decoding.py", line 201, in decode_choice_class
    return decode_dataclass(subcls, raw_value, path)
  File "/home/binliu/miniconda3/envs/lerobot/lib/python3.10/site-packages/draccus/parsers/decoding.py", line 159, in decode_dataclass
    instance = origin(**init_args)
  File "<string>", line 23, in __init__
  File "/home/binliu/lerobot/lerobot/common/robot_devices/control_configs.py", line 90, in __post_init__
    self.policy = PreTrainedConfig.from_pretrained(policy_path, cli_overrides=cli_overrides)
  File "/home/binliu/lerobot/lerobot/configs/policies.py", line 145, in from_pretrained
    return draccus.parse(cls, config_file, args=cli_overrides)
  File "/home/binliu/miniconda3/envs/lerobot/lib/python3.10/site-packages/draccus/argparsing.py", line 211, in parse
    return parser.parse_args(args)
  File "/home/binliu/miniconda3/envs/lerobot/lib/python3.10/site-packages/draccus/argparsing.py", line 102, in parse_args
    args, _ = self.parse_known_args(args, namespace, is_parse_args=True)
  File "/home/binliu/miniconda3/envs/lerobot/lib/python3.10/site-packages/draccus/argparsing.py", line 136, in parse_known_args
    parsed_t = self._postprocessing(parsed_args)
  File "/home/binliu/miniconda3/envs/lerobot/lib/python3.10/site-packages/draccus/argparsing.py", line 180, in _postprocessing
    cfg = decoding.decode(self.config_class, deflat_d)
  File "/home/binliu/miniconda3/envs/lerobot/lib/python3.10/site-packages/draccus/parsers/registry_utils.py", line 78, in wrapper
    return base_func(*args, **kw)
  File "/home/binliu/miniconda3/envs/lerobot/lib/python3.10/site-packages/draccus/parsers/decoding.py", line 48, in decode
    return get_decoding_fn(cls)(raw_value, ())  # type: ignore
  File "/home/binliu/miniconda3/envs/lerobot/lib/python3.10/site-packages/draccus/parsers/decoding.py", line 201, in decode_choice_class
    return decode_dataclass(subcls, raw_value, path)
  File "/home/binliu/miniconda3/envs/lerobot/lib/python3.10/site-packages/draccus/parsers/decoding.py", line 149, in decode_dataclass
    raise DecodingError(path, f"The fields {formatted_keys} are not valid for {stringify_type(cls)}")
draccus.utils.DecodingError: The fields `device`, `use_amp` are not valid for ACTConfig
```

Workaround:
1. Delete the following two items from the `config.json` of the `pretrained_model`:
```json
  "device": "cuda",
  "use_amp": false
```
2. Modify the `train_config.json` file of the `pretrained_model`. Move the two items `"device": "cuda"` and `"use_amp": false` out of the `"policy"` section and place them at the same level as the `"output_dir"` item.

### Expected behavior

Please help solve the problem.

johnny555, 2025-03-17:
I had a similar error trying to use the pi0 model. 

Turned out that I was on an old main branch. git pull managed to get rid of this for me.

imstevenpmwork, 2025-03-21:
@binliu-base Can you confirm this is happening in the latest main branch?