huggingface/lerobot #851: Hello, I would like to ask if I can use my ROS2 MoveIt2 robotic arm?

Gates-456, 2025-03-12:
Can it support ROS training? I believe this would be beneficial for ecosystem development.

gokulp01, 2025-03-12:
Hi @Gates-456 check out this repo: https://github.com/brukg/SO-100-arm I did get it working (with some changes) -- but overall this was a solid starting repository.

imstevenpmwork, 2025-03-14:
Hello @Gates-456, I'm not sure if I understood your question. Are you asking if:
1. You can use S0-100 with ROS2?
2. You can use lerobot training with the data from a SO-100 ROS2 enabled robot?

ycheng517, 2025-06-13:
I've added MoveIt2 support in: https://github.com/huggingface/lerobot/pull/1280

Gates-456, 2025-07-01:
> I've added MoveIt2 support in: [#1280](https://github.com/huggingface/lerobot/pull/1280)我已经添加了MoveIt2支持：[#1280](https://github.com/huggingface/lerobot/pull/1280)

Cool

ycheng517, 2025-08-04:
Since it looks like my PR won't be reviewed, I've created a repo for it and added some refinements, feel free to check it out: https://github.com/ycheng517/lerobot-ros