huggingface/lerobot #853: How to customize adding other robot and manipulator？

meijie-jesse, 2025-03-12:
Thanks for your great work! Now I got a problem how to customize adding other robot and manipulator. 

I have 7DOF bimanual manipulators robot, which is powered by servo-motor. I want to add it to lerobot so I can use this fantastic platform to collect data and train. Specially the ACT and diffusion policy.

I have the URDF file, and already setup in ROS moveit and Isaac Sim, using 485 to drive the real robot.

I checked the code and maybe I should crate new yaml file in /configs/robot an some other files for my robot.

Which is simpler compared to directly collecting data and training with ACT repository? Is there any tutorial on how to add a custom robot for fresh man?

Thanks a lot !

[image]

autmoate, 2025-03-12:
I would also be very interested in whether and how this project could be applied to other (6 DOF or, as the issuer suggests, even 7 DOF) robotic arms.
Given the differences in control systems, communication/control interfaces, and inverse kinematics, I assume this is certainly **not** trivial? Is there maybe some middleware solution or an abstraction layer that could facilitate adaptation?

What about 6 DOF robotic arms that are mechanically and kinematically similar to the SO-ARM100? Where would be a good starting point for adapting LeRobot? And what about simpler robotic arm designs, such as Arduino-based robots (e.g., Braccio)?

Thanks for this great project and models like Pi0—amazing work and truly exciting! I'm looking forward to testing it and look deeper into the code as soon as possible and hopefully adapting it to other robotic platforms. Alternatively, I might just have to build an SO-ARM100 pair...

ahmedtalbi, 2025-03-22:
I am also interested in this topic.

bjblank2, 2025-05-16:
I would also love some info on this topic. Working it from my own end, but I'm new to LeRobot and could always use some direction.

SakethTecholution, 2025-05-20:
yeah me too

Rip-Hunter, 2025-06-16:
Maybe this will be interesting to someone, here is a link to my fork Lerobot, into which my own environment and the VX300S robot were integrated, with modifications made for training with the pi0 model. The README describes the files to which the changes were made, I will write a little more detailed instructions later, but who needs it, please use and improve this solution, since it was created on the fly, and it doesn't look very good. The code has been commented for easier understanding.

https://github.com/Rip-Hunter/lerobot

autmoate, 2025-06-17:
Amazing @Rip-Hunter ! Thank you for sharing. I‘ll definitely have a look at it.

ycheng517, 2025-08-04:
I've created a lightweight ROS integration for LeRobot that should allow any ROS or MoveIt to be controlled by LeRobot. I've had success using this approach to collect teleoperation datasets and train models for my AR4 robot arm.

Feel free to check it out: https://github.com/ycheng517/lerobot-ros