huggingface/lerobot #860: Modify camera async_read/read API to return a dictionary instead of tuple for better compatability?

StoneT2000, 2025-03-13:
Currently the intel real sense camera api supports returning either a single rgb image or a rgb image and depth image as a 2-uple

https://github.com/huggingface/lerobot/blob/3c0a209f9fac4d2a57617e686a7f2a2309144ba2/lerobot/common/robot_devices/cameras/intelrealsense.py#L440-L443

However this is not super compatible to work with since not all cameras might return two values (open cv one only does rgb?). For a potentially better API would it be possible to have the async read / read functions always return a dictionary instead with some standard names and data types for the types of image data returned?

e.g.

```
return dict(rgb=..., depth=...)
```

This way it is also easier for me to check if the returned data has depth data or not.

Thanks!

CarolinePascal, 2025-04-14:
Hi @StoneT2000 !

Thank you for your suggestion regarding RGB and depth images ! Could you possibly fill a PR with the implemented `dict` solution so that we may discuss its integration with the other members of LeRobot team ?

Thanks !

Caroline.

CarolinePascal, 2025-05-26:
I'm closing this issue as depth images are no longer supported starting from #1154 .

Once again, sorry for the inconvenience :/

Best,

Caroline,

closed 2025-05-26 by CarolinePascal