huggingface/lerobot #881: [Bug] ImportError: cannot import name 'compute_stats' from 'lerobot.common.datasets.compute_stats'

yizt, 2025-03-20:
### System Info

```Shell
ubuntu linux
python 3.10.x
```

### Information

- [x] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

python lerobot/scripts/push_dataset_to_hub.py \
--raw-dir /path/to/raw dir \
--raw-format rm_pickle \
--push-to-hub 0 \
--local-dir /path/to/my local dir \
--repo-id lerobot/data set name

an import error occur
```
Traceback (most recent call last):
  File "/mnt/disk2/yizt/lerobot/lerobot/scripts/push_dataset_to_hub.py", line 56, in <module>
    from lerobot.common.datasets.compute_stats import compute_stats
ImportError: cannot import name 'compute_stats' from 'lerobot.common.datasets.compute_stats' (/mnt/disk2/yizt/lerobot/lerobot/common/datasets/compute_stats.py)
```

### Expected behavior

can you fix this bug,ths

imstevenpmwork, 2025-03-21:
Related to: https://github.com/huggingface/lerobot/pull/711#issuecomment-2690385750

imstevenpmwork, 2025-03-21:
`lerobot/scripts/push_dataset_to_hub.py` is indeed deprecated. Please use instead: https://github.com/huggingface/lerobot/blob/f39652707caed42a7cd5ab36066da5663b9565eb/lerobot/common/datasets/lerobot_dataset.py#L518

closed 2025-03-21 by imstevenpmwork