huggingface/lerobot #906: Code hets stuck at cv2.imshow() during teleoperation

SimoManni, 2025-03-26:
When I try to run the teleoperation or dataset recording with the camera, the code gets stuck while displaying the camera image. More specifically it freezes inside the `control_loop` function in `control_utils.py` at:

```python
        if display_cameras and not is_headless():
            image_keys = [key for key in observation if "image" in key]
            for key in image_keys:
                img = cv2.cvtColor(observation[key].numpy(), cv2.COLOR_RGB2BGR)
                print(key, img.shape, img.dtype)
                
                cv2.imwrite(f"{key}.png", img)
                cv2.imshow(key, img)
            
            cv2.waitKey(1)
```
The code never goes beyond the `cv2.imshow()`. As you can see from the code, I have tried to save the image and it works perfectly, meaning that there is no problem with the image itself or the camera. I have made sure to be running the script as the non-root user and that the environment is not headless. If I run a separate script to generate an image and display it using `cv2` it works perfectly, only within the provided scripts it fails. 

I am using the conda environment as suggested in the instructions on `Ubuntu 22.04`. The camera is an `Intel Realsense` but I don't think the problem is with the camera. Teleoperation and all other scripts are working correctly and if I remove the `cv2.imshow()` line the script sucessfully runs.

imstevenpmwork, 2025-03-27:
Please make sure you follow the recently updated instructions: https://github.com/huggingface/lerobot/blob/78fd1a1e04f87f3ec52428f72c29485124472fbc/README.md?plain=1#L95

For reference: 
* https://github.com/huggingface/lerobot/pull/883
* https://github.com/huggingface/lerobot/pull/907
* https://github.com/huggingface/lerobot/pull/911

DawnyWu, 2025-04-10:
I have the same problem.

imstevenpmwork, 2025-04-10:
We have dropped opencv show in favor of rerun, please use the latest main with a clean environment