huggingface/lerobot #912: When will MultiLeRobotDataset available?

Vacuame, 2025-03-28:
Hello, the MultiLeRobotDataset is very useful for training on large amounts of data; without it, training complex tasks would be difficult. However, I noticed that after the Simplify configs(#550) commit on January 31st,  MultiLeRobotDataset have been marked as unavailable(raise NotImplementedError("The MultiLeRobotDataset isn't supported for now.")). Could you please let me know approximately when this functionality will be restored, or why it has been made unavailable?

alexcbb, 2025-03-31:
Hi, I was also searching for a solution, it seems that there is the hacky solution from this PR that can help for now : https://github.com/huggingface/lerobot/pull/684

TangGuohh, 2025-04-01:
> Hello, the MultiLeRobotDataset is very useful for training on large amounts of data; without it, training complex tasks would be difficult. However, I noticed that after the Simplify configs([#550](https://github.com/huggingface/lerobot/pull/550)) commit on January 31st, MultiLeRobotDataset have been marked as unavailable(raise NotImplementedError("The MultiLeRobotDataset isn't supported for now.")). Could you please let me know approximately when this functionality will be restored, or why it has been made unavailable?您好，MultiLeRobotDataset 对于训练大量数据非常有用；如果没有它，训练复杂任务将会非常困难。但是，我注意到在 1 月 31 日提交简化配置 ( [#550](https://github.com/huggingface/lerobot/pull/550) ) 后，MultiLeRobotDataset 已被标记为不可用（引发 NotImplementedError（“目前不支持 MultiLeRobotDataset。”））。您能否告诉我大约何时恢复此功能，或者为什么它不可用？

Hello, I also struggled with this issue for a long time, but I've resolved it. The solution is efficient and verified to work correctly. You can use the merge.py script from my PR (#924) to solve your problem. I hope this helps you, and wish you all the best.

TangGuohh, 2025-04-01:
Now you only need to merge different datasets into a single directory, and it also supports padding of custom lengths.It automatically divides and organizes tasks, task_index, episodes, index, and frame_index.

Vacuame, 2025-04-05:
> Now you only need to merge different datasets into a single directory, and it also supports padding of custom lengths.It automatically divides and organizes tasks, task_index, episodes, index, and frame_index.

感谢分享(*^▽^*)，看来你是直接把两个数据集融合成一个，这种方法简单有效，而且可以长期使用，不受更新影响。在官方更新前这个是很好的替代方法