huggingface/lerobot #923: Cannot install Lerobot

Prasit7, 2025-03-31:
I am getting an error when the installation is building the av wheel. It is not passing this part of the installation

tuzhiLiu, 2025-03-31:
我直接运行`cd ~/lerobot && pip install -e ".[feetech]"`之后好像成功了

imstevenpmwork, 2025-03-31:
Please make sure you follow the recently updated instructions: https://github.com/huggingface/lerobot/blob/78fd1a1e04f87f3ec52428f72c29485124472fbc/README.md?plain=1#L95

For reference: 
* https://github.com/huggingface/lerobot/pull/883
* https://github.com/huggingface/lerobot/pull/907
* https://github.com/huggingface/lerobot/pull/911

sachaMorin, 2025-03-31:
Same issue here. I am using the updated instructions.

```
Failed to build av
ERROR: Failed to build installable wheels for some pyproject.toml based projects (av)
FATAL:   While performing build: while running engine: exit status 1
```

sachaMorin, 2025-03-31:
Fixed by running `conda install ffmpeg` even if ffmpeg was already available on the base machine.

Prasit7, 2025-03-31:
> Fixed by running `conda install ffmpeg` even if ffmpeg was already available on the base machine.

Did u run conda install ffmpeg right before running the installation?

sachaMorin, 2025-03-31:
Yes!

amisev, 2025-03-31:
Running into the same issue
```
 Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build editable ... done
  Preparing editable metadata (pyproject.toml) ... done
Collecting av<13.0.0,>=12.0.5 (from lerobot==0.1.0)
  Using cached av-12.3.0.tar.gz (3.8 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error
  
  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [29 lines of output]
      Package libavformat was not found in the pkg-config search path.
      Perhaps you should add the directory containing `libavformat.pc'
      to the PKG_CONFIG_PATH environment variable
      No package 'libavformat' found
      Package libavcodec was not found in the pkg-config search path.
      Perhaps you should add the directory containing `libavcodec.pc'
      to the PKG_CONFIG_PATH environment variable
      No package 'libavcodec' found
      Package libavdevice was not found in the pkg-config search path.
      Perhaps you should add the directory containing `libavdevice.pc'
      to the PKG_CONFIG_PATH environment variable
      No package 'libavdevice' found
      Package libavutil was not found in the pkg-config search path.
      Perhaps you should add the directory containing `libavutil.pc'
      to the PKG_CONFIG_PATH environment variable
      No package 'libavutil' found
      Package libavfilter was not found in the pkg-config search path.
      Perhaps you should add the directory containing `libavfilter.pc'
      to the PKG_CONFIG_PATH environment variable
      No package 'libavfilter' found
      Package libswscale was not found in the pkg-config search path.
      Perhaps you should add the directory containing `libswscale.pc'
      to the PKG_CONFIG_PATH environment variable
      No package 'libswscale' found
      Package libswresample was not found in the pkg-config search path.
      Perhaps you should add the directory containing `libswresample.pc'
      to the PKG_CONFIG_PATH environment variable
      No package 'libswresample' found
      pkg-config could not find libraries ['avformat', 'avcodec', 'avdevice', 'avutil', 'avfilter', 'swscale', 'swresample']
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
```

imstevenpmwork, 2025-04-01:
If you are in linux please run:

sudo apt-get install cmake build-essential python3-dev pkg-config libavformat-dev libavcodec-dev libavdevice-dev libavutil-dev libswscale-dev libswresample-dev libavfilter-dev pkg-config

amisev, 2025-04-01:
Thanks, I'm installing on Mac. I resolved an issue above ^^ by fixing PKG_CONFIG_PATH.
I got another one related to ffmpeg version, it turns out that av doesn't support ffmpeg 7 (see https://github.com/PyAV-Org/PyAV/issues/1463#issuecomment-2241603140). Will try to downgrade ffmpeg.

amisev, 2025-04-01:
`brew install ffmpeg@6` helped

Freeman656, 2025-04-01:
Hi,
I'm utilizing this thread to add some issue with the initial installation of av wheel. I'm getting this error :

Stored in directory: C:\Users\Freeman\AppData\Local\Temp\pip-ephem-wheel-cache-i9g5kfcf\wheels\c2\eb\98\7a2027992ac61335d7f804f57bf1bb076fb52ba50abed0d434
  Building wheel for av (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Building wheel for av (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [178 lines of output]
      C:\Users\Freeman\AppData\Local\Temp\pip-build-env-1nx3qvtv\overlay\Lib\site-packages\setuptools\_distutils\dist.py:289: UserWarning: Unknown distribution option: 'test_suite'
        warnings.warn(msg)
      C:\Users\Freeman\AppData\Local\Temp\pip-build-env-1nx3qvtv\overlay\Lib\site-packages\setuptools\dist.py:759: SetuptoolsDeprecationWarning: License classifiers are deprecated.

              ********************************************************************************
              Please consider removing the following classifiers in favor of a SPDX license expression:

              License :: OSI Approved :: BSD License

              See https://packaging.python.org/en/latest/guides/writing-pyproject-toml/#license for details.
              ********************************************************************************

      
        self._finalize_license_expression()
      running bdist_wheel

imstevenpmwork, 2025-04-01:
> Hi, I'm utilizing this thread to add some issue with the initial installation of av wheel. I'm getting this error :
>
> Stored in directory: [...]

This is a new one for me @Freeman656 Can you provide us more details? Can you share with us the output of the following commands?

```bash
uname -a
ffmpeg -version
ls /opt/homebrew/lib/libav* # for Mac
ldconfig -p | grep libav* # for Linux
```

In the virtual environment:
```bash
ffmpeg -version
pip freeze | grep torch
pip freeze | grep av
```

If you are in Windows, you will have to follow these instructions, although I haven't tested them: https://pyav.org/docs/develop/overview/installation.html#build-on-windows
Mind compiling the right version `av>=12.0.5,<13.0.0`.

Freeman656, 2025-04-01:
I managed to install av separatly using pip install -v "av==12.3.0". And now it's fine...

alexyyyander, 2025-04-02:
> I managed to install av separatly using pip install -v "av==12.3.0". And now it's fine... btw, I'm on Windows.

it works, thx a lot

Freeman656, 2025-04-02:
Next step, next error :)

Any idea on this one ? It's probably some simple python error, but my knowledge is very limited on this subject : 

(lerobot) C:\Users\Freeman\lerobot>python lerobot/scripts/configure_motor.py --port COM3 --brand feetech --model sts3215 --baudrate 1000000 --ID 1
Traceback (most recent call last):
  File "C:\Users\Freeman\lerobot\lerobot\scripts\configure_motor.py", line 176, in <module>
    configure_motor(args.port, args.brand, args.model, args.ID, args.baudrate)
  File "C:\Users\Freeman\lerobot\lerobot\scripts\configure_motor.py", line 60, in configure_motor
    motor_bus_config_cls, motor_bus_cls, model_baudrate_table, series_baudrate_table = get_motor_bus_cls(
  File "C:\Users\Freeman\lerobot\lerobot\scripts\configure_motor.py", line 34, in get_motor_bus_cls
    from lerobot.common.robot_devices.motors.configs import FeetechMotorsBusConfig
ModuleNotFoundError: No module named 'lerobot'

closed 2025-04-02 by imstevenpmwork

rakhimovv, 2025-04-07:
To install evdev, it helped me to run the following commands:

```
export CC=/usr/bin/gcc
export CXX=/usr/bin/g++
```

fbeck-dev, 2025-04-07:
> I managed to install av separatly using pip install -v "av==12.3.0". And now it's fine... btw, I'm on Windows.

It also worked for me on mac os m series