huggingface/lerobot #933: alueError: One or several timestamps unexpectedly violate the tolerance inside episode range. This might be due to synchronization issues with timestamps during data collection.

GrantHen, 2025-04-03:
### System Info

```Shell
- `lerobot` version: 0.1.0
- Platform: Linux-6.11.0-21-generic-x86_64-with-glibc2.39
- Python version: 3.10.13
- Huggingface_hub version: 0.29.3
- Dataset version: 3.5.0
- Numpy version: 2.2.4
- PyTorch version (GPU?): 2.5.1+cu124 (True)
- Cuda version: 12040
- Using GPU in script?: 4070
```

### Information

- [ ] One of the scripts in the examples/ folder of LeRobot
- [x] My own task or dataset (give details below)

### Reproduction

When running this prompt to resume my progress

bash```
python lerobot/scripts/control_robot.py record --robot-path lerobot/configs/robot/so100.yaml --fps 30 --repo-id therarelab/so100_pick_place --tags so100 tutorial --warmup-time-s 5 --episode-time-s 30 --reset-time-s 10 --num-episodes 14 --push-to-hub 1 --single-task "Pick a plaster roll and place" --resume 1 
```
I get this error:
bash```
tasks.jsonl: 100%|████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 59.0/59.0 [00:00<00:00, 269kB/s]
stats.json: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 5.43k/5.43k [00:00<00:00, 16.9MB/s]
info.json: 100%|███████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 3.23k/3.23k [00:00<00:00, 14.6MB/s]
episodes.jsonl: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 881/881 [00:00<00:00, 4.62MB/s]
Fetching 4 files: 100%|███████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 4/4 [00:00<00:00, 14.10it/s]
.gitattributes: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 2.46k/2.46k [00:00<00:00, 11.7MB/s]
README.md: 100%|███████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 3.74k/3.74k [00:00<00:00, 19.1MB/s]
episode_000000.parquet: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 20.6k/20.6k [00:00<00:00, 3.36MB/s]
episode_000005.parquet: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 21.8k/21.8k [00:00<00:00, 3.65MB/s]
episode_000001.parquet: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 21.9k/21.9k [00:00<00:00, 9.13MB/s]
episode_000004.parquet: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 21.2k/21.2k [00:00<00:00, 4.67MB/s]
episode_000002.parquet: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 19.1k/19.1k [00:00<00:00, 6.31MB/s]
episode_000003.parquet: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 18.7k/18.7k [00:00<00:00, 8.60MB/s]
episode_000007.parquet: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 26.8k/26.8k [00:00<00:00, 69.1MB/s]
episode_000006.parquet: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 22.1k/22.1k [00:00<00:00, 16.8MB/s]
episode_000011.parquet: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 23.7k/23.7k [00:00<00:00, 56.8MB/s]
episode_000013.parquet: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 23.9k/23.9k [00:00<00:00, 65.7MB/s]
episode_000012.parquet: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 24.7k/24.7k [00:00<00:00, 23.7MB/s]
episode_000014.parquet: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 24.8k/24.8k [00:00<00:00, 64.1MB/s]
episode_000010.parquet: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 21.3k/21.3k [00:00<00:00, 54.8MB/s]
episode_000008.parquet: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 19.5k/19.5k [00:00<00:00, 50.7MB/s]
episode_000009.parquet: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 21.2k/21.2k [00:00<00:00, 61.9MB/s]
episode_000000.mp4: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 3.75M/3.75M [00:00<00:00, 14.0MB/s]
episode_000003.mp4: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 3.10M/3.10M [00:00<00:00, 7.06MB/s]
episode_000006.mp4: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 3.85M/3.85M [00:00<00:00, 7.31MB/s]
episode_000008.mp4: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 3.40M/3.40M [00:00<00:00, 17.2MB/s]
episode_000002.mp4: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 3.46M/3.46M [00:00<00:00, 5.54MB/s]
episode_000004.mp4: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 4.04M/4.04M [00:00<00:00, 5.87MB/s]
episode_000005.mp4: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 3.90M/3.90M [00:00<00:00, 5.17MB/s]
episode_000001.mp4: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 4.84M/4.84M [00:00<00:00, 6.47MB/s]
episode_000011.mp4: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 2.30M/2.30M [00:00<00:00, 19.2MB/s]
episode_000007.mp4: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 4.97M/4.97M [00:00<00:00, 6.00MB/s]
episode_000012.mp4: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 2.57M/2.57M [00:00<00:00, 8.29MB/s]
episode_000010.mp4: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 3.73M/3.73M [00:00<00:00, 11.7MB/s]
episode_000009.mp4: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 4.08M/4.08M [00:00<00:00, 10.2MB/s]
episode_000013.mp4: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 2.36M/2.36M [00:00<00:00, 9.01MB/s]
episode_000014.mp4: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 2.52M/2.52M [00:00<00:00, 6.85MB/s]
episode_000001.mp4: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 7.54M/7.54M [00:00<00:00, 23.6MB/s]
episode_000000.mp4: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 5.75M/5.75M [00:00<00:00, 7.26MB/s]
episode_000008.mp4: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 4.70M/4.70M [00:00<00:00, 24.1MB/s]
episode_000004.mp4: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 6.11M/6.11M [00:00<00:00, 9.72MB/s]
[... cut ...]
  --warmup-time-s 5 \
  --episode-time-s 30 \
  --reset-time-s 10 \
  --num-episodes 14 \
  --push-to-hub 1 \
  --single-task "Pick a plaster roll and place" \
  --resume 1 \
  --tolerance-s 0.5  # Increased tolerance to 0.5 seconds
usage: control_robot.py [-h] {calibrate,teleoperate,record,replay} ...
control_robot.py: error: unrecognized arguments: --tolerance-s 0.5
```

I've tried changing every variable.

### Expected behavior

Expected behavior is the episodes would run correctly and add to the repository.

qxiao0521, 2025-04-24:
Hi!Have you solved this problem?I've encountered the same problem and haven't found a suitable solution yet.Can you give me some advice?

r7vme, 2025-05-10:
For me the problem was related to invalid dataset. I had 907 complete episodes, but some garbage samples after. Check that your last sample in episode actually last sample in the dataset

imstevenpmwork, 2026-03-11:
Hello!

I'm closing this ticket as I believe multiple fixes related to this were shipped to `main`, plus `tolerance_s` is now configurable from the CLI: https://github.com/huggingface/lerobot/pull/2653
This however, might not be a proper fix for the underlying issue, so feel free to reopen if you believe this continues to be a problem in latest `main`

Duplicate of: https://github.com/huggingface/lerobot/issues/1875

Thanks!