huggingface/lerobot #941: New ffmpeg does not work with PyAV

gu021, 2025-04-06:
OS: Ubuntu  22.0.4
python:  3.10
virtual env: miniconda
ffmpeg:  7.1.1
open CV:  4.11
av == 14.2.0
torchvision == 0.21.0
av and torchvision were default. After update to above version, they gave the same result.

Issue: In order to record video, newest FFmpeg is a must since certain encoder is used. When I do the following command:

pip install --no-binary=av -e ".[feetech]"

or 

pip install --no-binary=av -e .

Terminal gave the same output as follow, which indicate: " error: ‘AV_OPT_TYPE_CHANNEL_LAYOUT’ undeclared (first use in this function); did you mean ‘AV_OPT_TYPE_CHLAYOUT’? "
"
Possible reason is: FFmpeg changed   AV_OPT_TYPE_CHANNEL_LAYOUT   into  AV_OPT_TYPE_CHLAYOUT in new version.  It is not compatible in PyAV. 

Suggetion: can we have more control of versions by writing an requirement.txt indicating import package versions in order to avoid conflicts. 

The following paragraphs are the terminal outputs: 

      building 'av.format' extension
      gcc -pthread -B /home/guyu/miniconda3/envs/lerobot/compiler_compat -Wno-unused-result -Wsign-compare -DNDEBUG -fwrapv -O2 -Wall -fPIC -O2 -isystem /home/guyu/miniconda3/envs/lerobot/include -fPIC -O2 -isystem /home/guyu/miniconda3/envs/lerobot/include -fPIC -I/usr/include/x86_64-linux-gnu -I/home/guyu/miniconda3/envs/lerobot/include/python3.10 -c src/av/format.c -o build/temp.linux-x86_64-cpython-310/src/av/format.o
      src/av/format.c: In function ‘__pyx_pf_2av_6format_15ContainerFormat___cinit__’:
      src/av/format.c:3373:24: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
       3373 |     __pyx_v_self->iptr = av_find_input_format(__pyx_t_4);
            |                        ^
      src/av/format.c:3410:24: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
       3410 |     __pyx_v_self->optr = av_guess_format(__pyx_t_5, NULL, NULL);
            |                        ^
      gcc -pthread -B /home/guyu/miniconda3/envs/lerobot/compiler_compat -shared -Wl,--allow-shlib-undefined -Wl,-rpath,/home/guyu/miniconda3/envs/lerobot/lib -Wl,-rpath-link,/home/guyu/miniconda3/envs/lerobot/lib -L/home/guyu/miniconda3/envs/lerobot/lib -Wl,--allow-shlib-undefined -Wl,-rpath,/home/guyu/miniconda3/envs/lerobot/lib -Wl,-rpath-link,/home/guyu/miniconda3/envs/lerobot/lib -L/home/guyu/miniconda3/envs/lerobot/lib build/temp.linux-x86_64-cpython-310/src/av/format.o -lavformat -lavcodec -lavdevice -lavutil -lavfilter -lswscale -lswresample -o build/lib.linux-x86_64-cpython-310/av/format.cpython-310-x86_64-linux-gnu.so
      building 'av.bytesource' extension
      gcc -pthread -B /home/guyu/miniconda3/envs/lerobot/compiler_compat -Wno-unused-result -Wsign-compare -DNDEBUG -fwrapv -O2 -Wall -fPIC -O2 -isystem /home/guyu/miniconda3/envs/lerobot/include -fPIC -O2 -isystem /home/guyu/miniconda3/envs/lerobot/include -fPIC -I/usr/include/x86_64-linux-gnu -I/home/guyu/miniconda3/envs/lerobot/include/python3.10 -c src/av/bytesource.c -o build/temp.linux-x86_64-cpython-310/src/av/bytesource.o
      gcc -pthread -B /home/guyu/miniconda3/envs/lerobot/compiler_compat -shared -Wl,--allow-shlib-undefined -Wl,-rpath,/home/guyu/miniconda3/envs/lerobot/lib -Wl,-rpath-link,/home/guyu/miniconda3/envs/lerobot/lib -L/home/guyu/miniconda3/envs/lerobot/lib -Wl,--allow-shlib-undefined -Wl,-rpath,/home/guyu/miniconda3/envs/lerobot/lib -Wl,-rpath-link,/home/guyu/miniconda3/envs/lerobot/lib -L/home/guyu/miniconda3/envs/lerobot/lib build/temp.linux-x86_64-cpython-310/src/av/bytesource.o -lavformat -lavcodec -lavdevice -lavutil -lavfilter -lswscale -lswresample -o build/lib.linux-x86_64-cpython-310/av/bytesource.cpython-310-x86_64-linux-gnu.so
      building 'av.dictionary' extension
      gcc -pthread -B /home/guyu/miniconda3/envs/lerobot/compiler_compat -Wno-unused-result -Wsign-compare -DNDEBUG -fwrapv -O2 -Wall -fPIC -O2 -isystem /home/guyu/miniconda3/envs/lerobot/include -fPIC -O2 -isystem /home/guyu/miniconda3/envs/lerobot/include -fPIC -I/usr/include/x86_64-linux-gnu -I/home/guyu/miniconda3/envs/lerobot/include/python3.10 -c src/av/dictionary.c -o build/temp.linux-x86_64-cpython-310/src/av/dictionary.o
      gcc -pthread -B /home/guyu/miniconda3/envs/lerobot/compiler_compat -shared -Wl,--allow-shlib-undefined -Wl,-rpath,/home/guyu/miniconda3/envs/lerobot/lib -Wl,-rpath-link,/home/guyu/miniconda3/envs/lerobot/lib -L/home/guyu/miniconda3/envs/lerobot/lib -Wl,--allow-shlib-undefined -Wl,-rpath,/home/guyu/miniconda3/envs/lerobot/lib -Wl,-rpath-link,/home/guyu/miniconda3/envs/lerobot/lib -L/home/guyu/miniconda3/envs/lerobot/lib build/temp.linux-x86_64-cpython-310/src/av/dictionary.o -lavformat -lavcodec -lavdevice -lavutil -lavfilter -lswscale -lswresample -o build/lib.linux-x86_64-cpython-310/av/dictionary.cpython-310-x86_64-linux-gnu.so
      building 'av.option' extension
      gcc -pthread -B /home/guyu/miniconda3/envs/lerobot/compiler_compat -Wno-unused-result -Wsign-compare -DNDEBUG -fwrapv -O2 -Wall -fPIC -O2 -isystem /home/guyu/miniconda3/envs/lerobot/include -fPIC -O2 -isystem /home/guyu/miniconda3/envs/lerobot/include -fPIC -I/usr/include/x86_64-linux-gnu -I/home/guyu/miniconda3/envs/lerobot/include/python3.10 -c src/av/option.c -o build/temp.linux-x86_64-cpython-310/src/av/option.o
      src/av/option.c: In function ‘__pyx_pymod_exec_option’:
      src/av/option.c:7416:51: error: ‘AV_OPT_TYPE_CHANNEL_LAYOUT’ undeclared (first use in this function); did you mean ‘AV_OPT_TYPE_CHLAYOUT’?
       7416 |   __pyx_t_3 = __Pyx_PyInt_From_enum__AVOptionType(AV_OPT_TYPE_CHANNEL_LAYOUT); if (unlikely(!__pyx_t_3)) __PYX_ERR(0, 36, __pyx_L1_error)
            |                                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~
            |                                                   AV_OPT_TYPE_CHLAYOUT
      src/av/option.c:7416:51: note: each undeclared identifier is reported only once for each function it appears in
      error: command '/usr/bin/gcc' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for av
Successfully built lerobot
Failed to build av
ERROR: Failed to build installable wheels for some pyproject.toml based projects (av)
(lerobot) guyu@GuLinux:~/Documents/5_lerobot/lerobot$ 

> What you should do according to the new update is the following:
> 
> 1. **Fetch the latest changes** (if you haven't already done that).
> 
> 2. **Create a new virtual environment using conda**:
>    ```bash
>    conda create -n lerobot python=3.10 -y
>    conda activate lerobot
>    ```
> 
> 3. **Install ffmpeg using conda**. This is in accordance with the README, just a few additions to ensures ffmpeg version ≥ 7:
>    ```bash
>    conda install -c conda-forge 'ffmpeg>=7.0' -y
>    ```
> 
> 4. **Install lerobot** (change the extras based on the robot you're using; this one is for Aloha):
>    ```bash
>    pip install --no-binary=av -e ".[dynamixel, intelrealsense]"
>    ```
> 
> These steps should be enough to avoid any issues with encoders or `jpeg12_write_raw_data`. 

 _Originally posted by @shantanuparab-tr in [#679](https://github.com/huggingface/lerobot/issues/679#issuecomment-2773418429)_

Tavish9, 2025-04-07:
follow my suggestions in issue #727 :)

imstevenpmwork, 2025-04-07:
I don't understand the meaning of this issue. Did you follow the recommended instructions for installation?

Download our source code:
```bash
git clone https://github.com/huggingface/lerobot.git
cd lerobot
```

Create a virtual environment with Python 3.10 and activate it, e.g. with [`miniconda`](https://docs.anaconda.com/free/miniconda/index.html):
```bash
conda create -y -n lerobot python=3.10
conda activate lerobot
```

When using `miniconda`, if you don't have `fffmpeg` in your environment:
```bash
conda install ffmpeg
```

Install 🤗 LeRobot:
```bash
pip install --no-binary=av -e .
```

> **NOTE:** If you encounter build errors, you may need to install additional dependencies (`cmake`, `build-essential`, and `ffmpeg libs`). On Linux, run:
`sudo apt-get install cmake build-essential python-dev pkg-config libavformat-dev libavcodec-dev libavdevice-dev libavutil-dev libswscale-dev libswresample-dev libavfilter-dev pkg-config`. For other systems, see: [Compiling PyAV](https://pyav.org/docs/develop/overview/installation.html#bring-your-own-ffmpeg)

This should install ffmpeg 6.1.1. The issue of having ffmpeg 7 is that it is not compatible with the version packaged by opencv

shantanuparab-tr, 2025-04-07:
Here's what I have noticed.  Just running 
```bash 
conda install ffmpeg
``` 
installs `ffmpeg` version `6.1.1`. This version does not support `libsvtav1` encoder

```bash
ffmpeg version 6.1.1 Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 11.2.0 (Anaconda gcc)
  configuration: --prefix=/home/interbotix/miniconda3/envs/ffmpeg_test --cc=/croot/ffmpeg_1743153284778/_build_env/bin/x86_64-conda-linux-gnu-cc --ar=/croot/ffmpeg_1743153284778/_build_env/bin/x86_64-conda-linux-gnu-ar --nm=/croot/ffmpeg_1743153284778/_build_env/bin/x86_64-conda-linux-gnu-nm --ranlib=/croot/ffmpeg_1743153284778/_build_env/bin/x86_64-conda-linux-gnu-ranlib --strip=/croot/ffmpeg_1743153284778/_build_env/bin/x86_64-conda-linux-gnu-strip --disable-doc --enable-swresample --enable-swscale --enable-openssl --enable-libxml2 --enable-libtheora --enable-demuxer=dash --enable-postproc --enable-hardcoded-tables --enable-libfreetype --enable-libharfbuzz --enable-libfontconfig --enable-libdav1d --enable-zlib --enable-libaom --enable-pic --enable-shared --disable-static --disable-gpl --enable-version3 --disable-sdl2 --enable-libopenh264 --enable-libopus --enable-libmp3lame --enable-libopenjpeg --enable-libvorbis --enable-pthreads --enable-libtesseract --enable-libvpx
  libavutil      58. 29.100 / 58. 29.100
  libavcodec     60. 31.102 / 60. 31.102
  libavformat    60. 16.100 / 60. 16.100
  libavdevice    60.  3.100 / 60.  3.100
  libavfilter     9. 12.100 /  9. 12.100
  libswscale      7.  5.100 /  7.  5.100
  libswresample   4. 12.100 /  4. 12.100
 V....D libaom-av1           libaom AV1 (codec av1)
 V....D libopenh264          OpenH264 H.264 / AVC / MPEG-4 AVC / MPEG-4 part 10 (codec h264)
 VF.... libopenjpeg          OpenJPEG JPEG 2000 (codec jpeg2000)
 V....D libtheora            libtheora Theora (codec theora)
 V....D libvpx               libvpx VP8 (codec vp8)
 V....D libvpx-vp9           libvpx VP9 (codec vp9)
 VF...D zlib                 LCL (LossLess Codec Library) ZLIB
 A....D libmp3lame           libmp3lame MP3 (MPEG audio layer 3) (codec mp3)
 A....D libopus              libopus Opus (codec opus)
 A....D libvorbis            libvorbis (codec vorbis)
```

Installing ffmpeg using ```bash conda install -c conda-forge 'ffmpeg>=7.0' -y```  gives you all the required encoders.

```bash
ffmpeg version 7.1.1 Copyright (c) 2000-2025 the FFmpeg developers
  built with gcc 13.3.0 (conda-forge gcc 13.3.0-2)
  configuration: --prefix=/home/interbotix/miniconda3/envs/lerobot_trossen_env --cc=/home/conda/feedstock_root/build_artifacts/ffmpeg_1743376049581/_build_env/bin/x86_64-conda-linux-gnu-cc --cxx=/home/conda/feedstock_root/build_artifacts/ffmpeg_1743376049581/_build_env/bin/x86_64-conda-linux-gnu-c++ --nm=/home/conda/feedstock_root/build_artifacts/ffmpeg_1743376049581/_build_env/bin/x86_64-conda-linux-gnu-nm --ar=/home/conda/feedstock_root/build_artifacts/ffmpeg_1743376049581/_build_env/bin/x86_64-conda-linux-gnu-ar --disable-doc --enable-openssl --enable-demuxer=dash --enable-hardcoded-tables --enable-libfreetype --enable-libharfbuzz --enable-libfontconfig --enable-libopenh264 --enable-libdav1d --disable-gnutls --enable-libvpx --enable-libass --enable-pthreads --enable-alsa --enable-libpulse --enable-vaapi --enable-libopenvino --enable-gpl --enable-libx264 --enable-libx265 --enable-libmp3lame --enable-libaom --enable-libsvtav1 --enable-libxml2 --enable-pic --enable-shared --disable-static --enable-version3 --enable-zlib --enable-libvorbis --enable-libopus --enable-librsvg --enable-ffplay --pkg-config=/home/conda/feedstock_root/build_artifacts/ffmpeg_1743376049581/_build_env/bin/pkg-config
  libavutil      59. 39.100 / 59. 39.100
  libavcodec     61. 19.101 / 61. 19.101
  libavformat    61.  7.100 / 61.  7.100
  libavdevice    61.  3.100 / 61.  3.100
  libavfilter    10.  4.100 / 10.  4.100
  libswscale      8.  3.100 /  8.  3.100
  libswresample   5.  3.100 /  5.  3.100
  libpostproc    58.  3.100 / 58.  3.100
 V....D libaom-av1           libaom AV1 (codec av1)
 V..... libsvtav1            SVT-AV1(Scalable Video Technology for AV1) encoder (codec av1)
 V....D libx264              libx264 H.264 / AVC / MPEG-4 AVC / MPEG-4 part 10 (codec h264)
 V....D libx264rgb           libx264 H.264 / AVC / MPEG-4 AVC / MPEG-4 part 10 RGB (codec h264)
 V....D libopenh264          OpenH264 H.264 / AVC / MPEG-4 AVC / MPEG-4 part 10 (codec h264)
 V....D libx265              libx265 H.265 / HEVC (codec hevc)
 V....D libvpx               libvpx VP8 (codec vp8)
 V....D libvpx-vp9           libvpx VP9 (codec vp9)
 VF...D zlib                 LCL (LossLess Codec Library) ZLIB
 A....D libmp3lame           libmp3lame MP3 (MPEG audio layer 3) (codec mp3)
 A....D libopus              libopus Opus (codec opus)
 A....D libvorbis            libvorbis (codec vorbis)
```

Using ffmpeg 6.1.1 results in encoder not found issue as mentioned in 
1. #822 
2. #705 

I'm using Linux 22.04 , and I don't face any issues with pyav installations as I have downloaded all the required packages as mentioned in the Readme.

I hope this information helps. @imstevenpmwork

BruceGeLi, 2025-04-07:
Hi, I am using Ubuntu 22.04, and I also faced the same av build issue of using ffmpeg==7.1.1. 
Downgrading to 6.1.1 helps the installation. 

I am new to this repository, so I haven't run anything to face the encoder issue yet.

imstevenpmwork, 2025-04-07:
Hi @shantanuparab-tr,

Thank you for the detailed explanation. It looks like this is an additional challenge on top of the previous dependency version conflicts and the `imshow()` issue we encountered.

After checking, I confirm that the conda-installed FFmpeg doesn’t include `libsvtav1`, but the Brew-installed version (on macOS) does. While upgrading to FFmpeg 7 would include `libsvtav1`, it would reintroduce conflicts with OpenCV, leading to the same `imshow()` problem as before.

Interestingly, on Ubuntu 24, installing FFmpeg via `sudo apt install ffmpeg` provides version 6.1.1—which is compatible with our codebase—and includes `libsvtav1`. This suggests we could update the installation instructions to recommend installing FFmpeg through Brew (macOS) or APT (Ubuntu 24) instead of Conda, ensuring compatibility while keeping encoder support.

One caveat: Ubuntu 22’s default APT repository installs FFmpeg 4.X, so we’d need separate instructions for that version.
Another solution would be to build manually FFmpeg with all the blows and whistles and then point all the packages to the built version. But I'm afraid that would complicate the installation for an entry user.

gu021, 2025-04-08:
This is really a nice suggestion:

>  This suggests we could update the installation instructions to recommend installing FFmpeg through Brew (macOS) or APT (Ubuntu 24) instead of Conda, ensuring compatibility while keeping encoder support. ([#950](https://github.com/huggingface/lerobot/pull/950))

We need at least one working Ubuntu solution since Pi0 project is based on Ubuntu only. May I ask who will do the verification of above solution and do the update of installation instruction? A nice and reliable start is essential to a good result.

shantanuparab-tr, 2025-04-08:
Here's an update, I tried a few things and noticed that the order of installation really matters.

Here's the workflow that works for me on Linux 22.04.

1. Create a conda virtual environment.
```bash
conda create -n lerobot python=3.10 -y
conda activate lerobot
```
2. Installing LeRobot
```bash
pip install --no-binary=av -e ".[dynamixel, intelrealsense]"
```
3. Installing `ffmpeg`
```bash
conda install -c conda-forge 'ffmpeg>=7.0' -y
```

How this help, this allows you to build pyav from source without getting this error
```bash
  src/av/option.c: In function ‘__pyx_pymod_exec_option’:
  src/av/option.c:7416:51: error: ‘AV_OPT_TYPE_CHANNEL_LAYOUT’ undeclared (first use in this function); did you mean ‘AV_OPT_TYPE_CHLAYOUT’?
   7416 |   __pyx_t_3 = __Pyx_PyInt_From_enum__AVOptionType(AV_OPT_TYPE_CHANNEL_LAYOUT); if (unlikely(!__pyx_t_3)) __PYX_ERR(0, 36, __pyx_L1_error)
        |                                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~
        |                                                   AV_OPT_TYPE_CHLAYOUT
  src/av/option.c:7416:51: note: each undeclared identifier is reported only once for each function it appears in
  error: command '/usr/bin/gcc' failed with exit code 1
  [end of output]
```

Installing ffmpeg later allows you to get the correct version of ffmpeg that supports `libsvtav1`

Note: Here's why the previous solution worked Isuccessfully build pyav from source in one of my environments, whihc was then cached, this cached version was used in the other environments as well, therefore, I didn't run into the pyav not built error.

When I used ```pip install --no-binary=av -e . --no-cache-dir --force-reinstall``` I was able to recreate the issue.  This led me to the above solution.

gu021, 2025-04-08:
The following suggestion is really nice This suggests we could update the installation instructions to recommend installing FFmpeg through Brew (macOS) or APT (Ubuntu 24) instead of Conda, ensuring compatibility while keeping encoder support. (#950)

gu021, 2025-04-08:
Nice solution! I believe this time it will work. You found the root cause. I will try it out soon. Thanks a lot, Shantanuparab!

imstevenpmwork, 2025-04-09:
We're working on a more elaborated solution, I recommend you to give first this PR a try (from a fresh environment and following the new installation instructions in the main readme): https://github.com/huggingface/lerobot/pull/903

shrutichakraborty, 2025-04-09:
Hello! I created an issue #964 as I have having some related issues. Using the fixes provided by @shantanuparab-tr I had done `conda install -c conda-forge 'ffmpeg>=7.0' -y` and since then my installation worked fine, I was able to run the `control_robot.py` script perfectly and collect a dataset. Now when I tried to train I had an issue where I get : 

```
NFO 2025-04-09 16:58:03 ts/train.py:202 Start offline training on a fixed dataset
Traceback (most recent call last):
  File "/home/moonshot/lerobot/lerobot/scripts/train.py", line 288, in <module>
    train()
  File "/home/moonshot/lerobot/lerobot/configs/parser.py", line 227, in wrapper_inner
    response = fn(cfg, *args, **kwargs)
  File "/home/moonshot/lerobot/lerobot/scripts/train.py", line 205, in train
    batch = next(dl_iter)
  File "/home/moonshot/lerobot/lerobot/common/datasets/utils.py", line 588, in cycle
    yield next(iterator)
  File "/home/moonshot/miniconda3/envs/lerobot/lib/python3.10/site-packages/torch/utils/data/dataloader.py", line 701, in __next__
    data = self._next_data()
  File "/home/moonshot/miniconda3/envs/lerobot/lib/python3.10/site-packages/torch/utils/data/dataloader.py", line 1465, in _next_data
    return self._process_data(data)
  File "/home/moonshot/miniconda3/envs/lerobot/lib/python3.10/site-packages/torch/utils/data/dataloader.py", line 1491, in _process_data
    data.reraise()
  File "/home/moonshot/miniconda3/envs/lerobot/lib/python3.10/site-packages/torch/_utils.py", line 715, in reraise
    raise exception
RuntimeError: Caught RuntimeError in DataLoader worker process 0.
Original Traceback (most recent call last):
  File "/home/moonshot/miniconda3/envs/lerobot/lib/python3.10/site-packages/torch/utils/data/_utils/worker.py", line 351, in _worker_loop
    data = fetcher.fetch(index)  # type: ignore[possibly-undefined]
  File "/home/moonshot/miniconda3/envs/lerobot/lib/python3.10/site-packages/torch/utils/data/_utils/fetch.py", line 52, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/moonshot/miniconda3/envs/lerobot/lib/python3.10/site-packages/torch/utils/data/_utils/fetch.py", line 52, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/moonshot/lerobot/lerobot/common/datasets/lerobot_dataset.py", line 739, in __getitem__
    video_frames = self._query_videos(query_timestamps, ep_idx)
  File "/home/moonshot/lerobot/lerobot/common/datasets/lerobot_dataset.py", line 711, in _query_videos
    frames = decode_video_frames(video_path, query_ts, self.tolerance_s, self.video_backend)
  File "/home/moonshot/lerobot/lerobot/common/datasets/video_utils.py", line 66, in decode_video_frames
    return decode_video_frames_torchcodec(video_path, timestamps, tolerance_s)
  File "/home/moonshot/lerobot/lerobot/common/datasets/video_utils.py", line 190, in decode_video_frames_torchcodec
    from torchcodec.decoders import VideoDecoder
  File "/home/moonshot/miniconda3/envs/lerobot/lib/python3.10/site-packages/torchcodec/__init__.py", line 10, in <module>
    from . import decoders, samplers  # noqa
  File "/home/moonshot/miniconda3/envs/lerobot/lib/python3.10/site-packages/torchcodec/decoders/__init__.py", line 7, in <module>
    from ._core import VideoStreamMetadata
  File "/home/moonshot/miniconda3/envs/lerobot/lib/python3.10/site-packages/torchcodec/decoders/_core/__init__.py", line 8, in <module>
    from ._metadata import (
  File "/home/moonshot/miniconda3/envs/lerobot/lib/python3.10/site-packages/torchcodec/decoders/_core/_metadata.py", line 15, in <module>
    from torchcodec.decoders._core.video_decoder_ops import (
  File "/home/moonshot/miniconda3/envs/lerobot/lib/python3.10/site-packages/torchcodec/decoders/_core/video_decoder_ops.py", line 59, in <module>
    load_torchcodec_extension()
  File "/home/moonshot/miniconda3/envs/lerobot/lib/python3.10/site-packages/torchcodec/decoders/_core/video_decoder_ops.py", line 44, in load_torchcodec_extension
    raise RuntimeError(
RuntimeError: Could not load libtorchcodec. Likely causes:
          1. FFmpeg is not properly installed in your environment. We support
             versions 4, 5, 6 and 7.
          2. The PyTorch version (2.5.1+cu124) is not compatible with
             this version of TorchCodec. Refer to the version compatibility
             table:
             https://github.com/pytorch/torchcodec?tab=readme-ov-file#installing-torchcodec.
          3. Another runtime dependency; see exceptions below.
        The following exceptions were raised as we tried to load libtorchcodec:
        
[start of libtorchcodec loading traceback]
libavutil.so.59: cannot open shared object file: No such file or directory
/home/moonshot/miniconda3/envs/lerobot/lib/python3.10/site-packages/torchcodec/libtorchcodec6.so: undefined symbol: _ZNK3c1011StorageImpl27throw_data_ptr_access_errorEv
libavutil.so.57: cannot open shared object file: No such file or directory
/home/moonshot/miniconda3/envs/lerobot/lib/python3.10/site-packages/torchcodec/libtorchcodec4.so: undefined symbol: _ZNK3c1011StorageImpl27throw_data_ptr_access_errorEv
[end of libtorchcodec loading traceback].

``` 

I have mentioned the versions for `ffmpeg` and `libtorchcodec` on #964, I then came across this issue and tried to downgrade `ffmpeg` to 6.1.1 and retry, but was not successful. Any suggestions? 

thanks

gu021, 2025-04-09:
I was testing against /feat/ branch. Then I noticed that it is brought to main. Then I tested main branch again. At least so far, I do not have any issue with FFmpeg version any more. I would like to close this issue. Thanks a lot, all!

closed 2025-04-09 by gu021