huggingface/lerobot #951: ACT model returns constant actions

kappa95, 2025-04-08:
### System Info

```Shell
Dear community,

I have a big issue with ACT model applied for teach a robot to shake a bottle in a specific way.
I have trained a real UR robot creating a LeRobotDataset containing observation.state equal to the current joint state, 2 realsense RGB pictures for every frame and action as the forward euler of the joint speed. 
I have tried both default values and different hyper params such as: kL = 100. LR: 5e-5. Batch size: 32. The problem is that in both cases the action is almost constant equal to the initial joint position (initial state).
```

### Information

- [ ] One of the scripts in the examples/ folder of LeRobot
- [x] My own task or dataset (give details below)

### Reproduction

Trained data and then infer using those features:
```
features={
            'observation.image.front_image': {
                'dtype': 'image',
                'shape': (frame_resolution_wrist[1], frame_resolution_wrist[0], 3),
                'names': ['height', 'width', 'channel']
            },
            'observation.image.side_image': {
                'dtype': 'image',
                'shape': (frame_resolution_world[1], frame_resolution_world[0], 3),
                'names': ['height', 'width', 'channel']
            },
            'observation.state': {
                'dtype': 'float32',
                'shape': (7, ),
                'names': ['position'],
            },
            'action': {
                'dtype': 'float32',
                'shape': (7, ),
                'names': ['position'],
            },
        },
```
and infer using:

for i in range(200, 400):
        sample = dataset[i]

        # Load and prepare inputs
        joint_state = sample['observation.state'].unsqueeze(0).to("cuda")  # [1, 7]
        pic_front = sample['observation.image.front_image'].unsqueeze(0).float().to("cuda") / 255.0  # [1, 3, 480, 640]
        pic_side = sample['observation.image.side_image'].unsqueeze(0).float().to("cuda") / 255.0    # [1, 3, 480, 640]

        input_data = {
            "observation.state": joint_state,
            "observation.image.front_image": pic_front,
            "observation.image.side_image": pic_side,
        }

        with torch.inference_mode():
            action_chunk = policy.select_action(input_data)  # returns [1, 100, 7]
        action_chunk_list.append(action_chunk.cpu().numpy())

### Expected behavior

Expect that the robot is effectively doing something. I tried using to infer some sample of the dataset but it stays constant.

Intellouis, 2025-04-11:
We have encountered the same problem with ACT. We wonder if you have tried to use other algorithms, e.g. diffusion policies, and whether they have the same problems.

KDeng98, 2025-06-29:
We have also encountered the same problem with ACT while training with different tasks and datasets. Would you mind sharing any insight about solving this problem?