huggingface/lerobot #952: PI0 Evaluation results on LIBERO

Zenobiaql, 2025-04-08:
When I try to finetune pi0 on [libero-goal](https://huggingface.co/datasets/IPEC-COMMUNITY/libero_object_no_noops_lerobot), the results is very low and the success rate is just 41.8%, which is far from the results 98.8% from [openpi](https://github.com/Physical-Intelligence/openpi/blob/main/examples/libero/README.md). The finetuning setting is as follows:
```
batch size: 16 (4GPUs)
learning rate: 2.5e-5
train_expert_only: false
freeze_vision_encoder: True
```

DelinQu, 2025-04-08:
[image]

My evaluation results with pi0. 

[image]

[image]

zongmianli, 2025-04-09:
@DelinQu It'd be great if you can provide the full training config (in a forked repo if possible)? Thanks!

lzl2040, 2025-04-09:
@DelinQu Is the evaluation code the same as [libero/main.py](https://github.com/Physical-Intelligence/openpi/blob/main/examples/libero/main.py)? I would really appreciate it if you could point out what modifications are made to the ```lerobot```  training code.

Zenobiaql, 2025-04-09:
> <img alt="Image" width="332" src="https://private-user-images.githubusercontent.com/60593268/431347641-c6403130-25c6-42d7-8d28-1a975fe84462.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDQxNzkxMzUsIm5iZiI6MTc0NDE3ODgzNSwicGF0aCI6Ii82MDU5MzI2OC80MzEzNDc2NDEtYzY0MDMxMzAtMjVjNi00MmQ3LThkMjgtMWE5NzVmZTg0NDYyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA0MDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNDA5VDA2MDcxNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTAzNDA2ZDRkZjQ1NzNkNTYyNjUwODM2M2Q0OTBiYTlhM2JlODk5ZDU5OTdiYjE3YjdjZjRmMGIzMDJkYzcwNzcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.j3XcHA1mrpsDMr0B8_TRZGXixqeJNC8Hjpw9VFcOxUo">
> 
> My evaluation results with pi0.
> <img alt="Image" width="2000" src="https://private-user-images.githubusercontent.com/60593268/431348550-ba5ecc4f-2620-4214-b230-097d4b499bd9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDQxNzkxMzUsIm5iZiI6MTc0NDE3ODgzNSwicGF0aCI6Ii82MDU5MzI2OC80MzEzNDg1NTAtYmE1ZWNjNGYtMjYyMC00MjE0LWIyMzAtMDk3ZDRiNDk5YmQ5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA0MDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNDA5VDA2MDcxNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWFlZjMyNWUyYTI4YjcxMjUzMTlkYzFmNzYzOTA0MDM5MTdhYTVkMzE3OTg0YmIyZGExYjE0OTE3MjViODFmNmEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0._Mx3ZlVhJA2PlYyasTpVSIuzDapOMxmfYJr8iLbU8IY"> <img alt="Image" width="804" src="https://private-user-images.githubusercontent.com/60593268/431348750-42f5b8c9-8592-40fc-b324-be3dc5a0d798.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDQxNzkxMzUsIm5iZiI6MTc0NDE3ODgzNSwicGF0aCI6Ii82MDU5MzI2OC80MzEzNDg3NTAtNDJmNWI4YzktODU5Mi00MGZjLWIzMjQtYmUzZGM1YTBkNzk4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA0MDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNDA5VDA2MDcxNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPThlNzkzOWUyNDFmNDYzZDhkNWMxNzIzMmRhZmI1ZGNjODE3MDkxZTUzYTlhZmI5ZDc1NzJkNmRmYjllZGZiZGImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.0hyd8_U4scb0-amGHJYDamGNWbEtUy0pN4IGSujbtvc">

The model is trained on the combination of four datasets from libero or just on single dataset?

DelinQu, 2025-04-09:
> The model is trained on the combination of four datasets from libero or just on single dataset?

The model is trained separately. But the performance is comparable if you trained in combination.

Zenobiaql, 2025-04-09:
Can you share the training code or the evaluation code? If could, I would appreciate it.

Zenobiaql, 2025-04-09:
> The model is trained separately. But the performance is comparable if you trained in combination.

Do you use the cosine scheduler for learning rate?

DelinQu, 2025-04-09:
> [@DelinQu](https://github.com/DelinQu) Is the evaluation code the same as [libero/main.py](https://github.com/Physical-Intelligence/openpi/blob/main/examples/libero/main.py)? I would really appreciate it if you could point out what modifications are made to the `lerobot` training code.

Yes! I just copied the code and adapted to lerobot inference style.

lzl2040, 2025-04-10:
> Hi! My code is adapted from the pr [#317](https://github.com/huggingface/lerobot/pull/317) with additional image augmentation. I will release the distributed training with large-scale data mixture in a few days after solving the pi0-fast performance bugs in [#955](https://github.com/huggingface/lerobot/issues/955). You could use this code to reproduce the results: [#317](https://github.com/huggingface/lerobot/pull/317)

Thanks for your reply. The training batch size is ```16*8=128```? Besides, why is the initial learning rate of the warm up close to 9e-4 in the picture? My initial learning rate is 0.

DelinQu, 2025-04-11:
> Thanks for your reply. The training batch size is `16*8=128`? Besides, why is the initial learning rate of the warm-up close to 9e-4 in the picture? My initial learning rate is 0.

Yes, the global bs=128. The initial lr is zero, and the following is my detailed hparams:
 
- the training curve corresponding to results reported in https://github.com/huggingface/lerobot/issues/952#issuecomment-2786119537

[image]

- params (all 4 subset use same setting)
```json
{
  "env": {
    "value": null
  },
  "eval": {
    "value": {
      "batch_size": 50,
      "n_episodes": 50,
      "use_async_envs": false
    }
  },
  "seed": {
    "value": 42
  },
  "steps": {
    "value": 240000
  },
  },
  "device": {
    "value": "cuda"
  },
  "policy": {
    "value": {
      "type": "pi_next",
      "num_steps": 10,
      "use_cache": true,
      "chunk_size": 50,
      "proj_width": 1024,
      "n_obs_steps": 1,
      "optimizer_lr": 0.0001,
      "empty_cameras": 0,
      "max_state_dim": 32,
      "optimizer_eps": 1e-8,
      "input_features": {
        "observation.state": {
          "type": "STATE",
          "shape": [
            8
          ]
        },
        "observation.images.image": {
          "type": "VISUAL",
          "shape": [
            3,
            256,
            256
          ]
        },
        "observation.images.wrist_image": {
          "type": "VISUAL",
          "shape": [
            3,
            256,
            256
          ]
        }
      },
      "max_action_dim": 32,
      "n_action_steps": 50,
      "optimizer_betas": [
[... 301 lines cut ...]
    "value": 0
  },
  "save_checkpoint": {
    "value": true
  },
  "policy_optimizer_lr": {
    "value": 0.0001
  },
  "freeze_vision_encoder": {
    "value": true
  },
  "checkpoints_total_limit": {
    "value": 2
  },
  "attention_implementation": {
    "value": "eager"
  },
  "pretrained_paligemma_path": {
    "value": null
  },
  "use_policy_training_preset": {
    "value": true
  }
}
```

lzl2040, 2025-04-13:
Thanks for your reply! Now my loss has decreased to 0.003. For evaluation, my code is as follows:
```python
def normalize_gripper_action(action, binarize=True):
    """
    Changes gripper action (last dimension of action vector) from [0,1] to [-1,+1].
    Necessary for some environments (not Bridge) because the dataset wrapper standardizes gripper actions to [0,1].
    Note that unlike the other action dimensions, the gripper action is not normalized to [-1,+1] by default by
    the dataset wrapper.

    Normalization formula: y = 2 * (x - orig_low) / (orig_high - orig_low) - 1
    """
    # Just normalize the last action to [-1,+1].
    orig_low, orig_high = 0.0, 1.0
    action[..., -1] = 2 * (action[..., -1] - orig_low) / (orig_high - orig_low) - 1

    if binarize:
        # Binarize to -1 or +1.
        action[..., -1] = np.sign(action[..., -1])

    return action

def invert_gripper_action(action):
    """
    Flips the sign of the gripper action (last dimension of action vector).
    This is necessary for some environments where -1 = open, +1 = close, since
    the RLDS dataloader aligns gripper actions such that 0 = close, 1 = open.
    """
    action[..., -1] = action[..., -1] * -1.0
    return action

def get_libero_image(obs, resize_size):
    # pi0
    resize_size = resize_size[0]
    img = np.ascontiguousarray(obs["agentview_image"][::-1, ::-1])
    img = image_tools.convert_to_uint8(
        image_tools.resize_with_pad(img, resize_size, resize_size)
    )
    wrist_img = np.ascontiguousarray(obs["robot0_eye_in_hand_image"][::-1, ::-1])
    wrist_img = image_tools.convert_to_uint8(
        image_tools.resize_with_pad(wrist_img, resize_size, resize_size)
    )
    return img, wrist_img

img, wrist_img = get_libero_image(obs, resize_size)
state = np.concatenate((obs["robot0_eef_pos"], quat2axisangle(obs["robot0_eef_quat"]), obs["robot0_gripper_qpos"]))

# Save preprocessed image for replay video
replay_images.append(img)

image = torch.from_numpy(img)
wrist_img = torch.from_numpy(wrist_img)
wrist_img = wrist_img.to(torch.float32) / 255 # transform to [0, 1]
wrist_img = wrist_img.permute(2, 0, 1)
wrist_img = wrist_img.unsqueeze(0).to(device)
state = torch.from_numpy(state)
state = state.to(torch.float32)
image = image.to(torch.float32) / 255 # transform to [0, 1]
image = image.permute(2, 0, 1)

state = state.unsqueeze(0).to(device)
image = image.unsqueeze(0).to(device)

# Prepare observations dict
observation = {
    "observation.image.primary": image,
    "observation.image.wrist": wrist_img,
    "state": state,
    "task": [task_description]
}
# Query model to get action
with torch.inference_mode():
    action = policy.select_action(observation, action_step=action_plan) # 1 7
numpy_action = action.squeeze(0).to("cpu").numpy()
# Normalize gripper action [0,1] -> [-1,+1] because the environment expects the latter
numpy_action = normalize_gripper_action(numpy_action, binarize=True)
numpy_action = invert_gripper_action(numpy_action) # for libero, -1=open, 1=close
print('==>action is', numpy_action)
# Execute action in environment
obs, reward, done, info = env.step(numpy_action.tolist())
```
There has any error?

ybu-lxd, 2025-04-27:
Hello, sorry to bother you. Which virtual environment of the project did you use to train and reason about the LIBERO dataset? I don't see the virtual environment in lerobot. Is it this one? https://github.com/Physical-Intelligence/openpi

FanZhang91, 2025-05-07:
@Zenobiaql Have you solved this problem？

SherlockZhang666, 2025-05-12:
Hi, I want to ask if anyone has succeeded in evaluating the pretrained model of diffusion policy (inside lerobot) in Libero? I've tried my best for a long time, but hasn't succeeded till now. I would really appreciate it if anyone could tell me how to put the initial observations in Libero environment into diffusion policy.

ducido, 2025-06-30:
@DelinQu Hi, thanks for your contribution, I appreciate that!!

This is my results when finetuning pi0 lerobot on LIBERO no_noops

[image]

Here is the training loss:

[image]

I use the same config above with some modified:
- I use checkpoint at 40k step
- Gradient accumulation= 2, bs=16 cause I only have 4 gpus (to keep global batchsize=128 for each update times)
- Don't add Rotation augmentation
- source code:  #1246 
- The line 333 in modeling_pi0.py: [line 333](https://github.com/huggingface/lerobot/blob/69901b9b6a2300914ca3de0ea14b6fa6e0203bd4/lerobot/common/policies/pi0/modeling_pi0.py#L333C9-L333C60)

```
losses = losses[:, :, : self.config.max_action_dim]
---> losses = losses[:, :, : self.config.action_feature.shape[0]]
```

I wanna ask what should I consider more to reproduce your results, or do you have any insights of error what I might encounter, thank you so much!

P/S: I also tried to add Rotation Augmentation but the results seem no difference

ducido, 2025-07-13:
> [@ducido](https://github.com/ducido) can you provide the command after you using the source code of [#1246](https://github.com/huggingface/lerobot/pull/1246)? thx a lot!

Yeah, this is my command to finetune with #1246 :

```
cp configs/policy_config/default.json /your/checkpoint/directory/config.json
CUDA_VISIBLE_DEVICES=0,1,2,3 accelerate launch --num_processes=4 --main_process_port 29400 lerobot/scripts/train_accelerate.py \
  --policy.path=/your/checkpoint/directory \
  --dataset.repo_id=IPEC-COMMUNITY/libero_10_no_noops_1.0.0_lerobot \
  --output_dir=outputs/train/$(date +%Y-%m-%d)/$(date +%H-%M-%S)_your_exp \
  --job_name=your_exp \
  --config_path=configs/libero_config/default.json \ 
  --batch_size=16 \
  --policy.gradient_accumulation_steps=2 \
  --save_freq=20000
```

JustinKai0527, 2025-07-13:
Hi @ducido, thanks a lot for sharing! 🙏
Just a quick question — I saw you copied default.json to config.json in the checkpoint directory. Where can I find that default.json file?

Also, just to make sure I got the steps right, is this the correct way to set things up?

```
git clone https://github.com/huggingface/lerobot.git
cd lerobot
# Checkout pull request #1246
git fetch origin pull/1246/head:pr-1246
git checkout pr-1246
```
And after that, I can run the training command you provided?

I'm still new to VLA, so I really appreciate your guidance. Thanks again!

chenkang455, 2025-07-19:
> [@DelinQu](https://github.com/DelinQu) Hi, thanks for your contribution, I appreciate that!!
> 
> This is my results when finetuning pi0 lerobot on LIBERO no_noops
> 
> [image]
> 
> Here is the training loss:
> 
> [image]
> 
> I use the same config above with some modified:
> 
> * I use checkpoint at 40k step
> * Gradient accumulation= 2, bs=16 cause I only have 4 gpus (to keep global batchsize=128 for each update times)
> * Don't add Rotation augmentation
> * source code:  [[WIP] Multi-gpus training with accelerate #1246](https://github.com/huggingface/lerobot/pull/1246)
> 
> I wanna ask what should I consider more to reproduce your results, or do you have any insights of error what I might encounter, thank you so much!
> 
> P/S: I also tried to add Rotation Augmentation but the results seem no difference

Hi, @ducido can you share your config or code for the eval? Like your action chunk setting and do you use the binarize griper setting?

chenkang455, 2025-07-20:
> IPEC-COMMUNITY/libero_10_no_noops_1.0.0_lerobot

Hi, your reproduced dataset is based on the IPEC-COMMUNITY/libero_10_no_noops_1.0.0_lerobot, this dataset is state-only?

ymy1946676292, 2025-08-05:
> > > Hi! My code is adapted from the pr [#317](https://github.com/huggingface/lerobot/pull/317) with additional image augmentation. I will release the distributed training with large-scale data mixture in a few days after solving the pi0-fast performance bugs in [#955](https://github.com/huggingface/lerobot/issues/955). You could use this code to reproduce the results: 嗨！我的代码改编自 pr，并增加了图像增强。解决#955 中的 pi0-fast 性能问题后，我将在几天内发布大规模数据混合的分布式训练。您可以使用此代码来复现结果：[#317](https://github.com/huggingface/lerobot/pull/317)
> > 
> > 
> > Thanks for your reply. The training batch size is `16*8=128`? Besides, why is the initial learning rate of the warm-up close to 9e-4 in the picture? My initial learning rate is 0.感谢你的回复。训练的批次大小是 `16*8=128` 吗？另外，为什么图中预热阶段的初始学习率接近 9e-4？我的初始学习率是 0。
> 
> Yes, the global bs=128. The initial lr is zero, and the following is my detailed hparams:好的，这是翻译后的文本： 是的，全局 bs=128。初始 lr 为零，以下是我的详细超参数：
> 
> * the training curve corresponding to results reported in [PI0 Evaluation results on LIBERO is very low #952 (comment)](https://github.com/huggingface/lerobot/issues/952#issuecomment-2786119537)与 [PI0 Evaluation results on LIBERO is very low #952 (comment)](https://github.com/huggingface/lerobot/issues/952#issuecomment-2786119537) 中报告的结果相对应的训练曲线
> 
> <img alt="Image" width="2000" src="https://private-user-images.githubusercontent.com/60593268/432575891-7764776d-9c5c-4f88-89cc-f92c88a1f1ad.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTQzODE4MzEsIm5iZiI6MTc1NDM4MTUzMSwicGF0aCI6Ii82MDU5MzI2OC80MzI1NzU4OTEtNzc2NDc3NmQtOWM1Yy00Zjg4LTg5Y2MtZjkyYzg4YTFmMWFkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA4MDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwODA1VDA4MTIxMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPThjOWRjZjVmMDlmOTY0MjU5YzcxYWZlYzQyZDJkODNkNTkwYWY1MGE4MWQyMGI3MGRlMDFhNWZmNzdmNDZkOWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.l7Ko0FOkhpODt6MGl502KBXp20tQ3aZe1YGQyhw9cpI">
> * params (all 4 subset use same setting)参数（所有 4 个子集使用相同设置）
> 
> {
>   "env": {
>     "value": null
>   },
>   "eval": {
>     "value": {
>       "batch_size": 50,
>       "n_episodes": 50,
>       "use_async_envs": false
>     }
>   },
>   "seed": {
>     "value": 42
>   },
>   "steps": {
>     "value": 240000
>   },
>   },
>   "device": {
>     "value": "cuda"
>   },
>   "policy": {
>     "value": {
>       "type": "pi_next",
>       "num_steps": 10,
>       "use_cache": true,
>       "chunk_size": 50,
>       "proj_width": 1024,
>       "n_obs_steps": 1,
>       "optimizer_lr": 0.0001,
>       "empty_cameras": 0,
>       "max_state_dim": 32,
>       "optimizer_eps": 1e-8,
>       "input_features": {
>         "observation.state": {
>           "type": "STATE",
>           "shape": [
>             8
>           ]
>         },
>         "observation.images.image": {
>           "type": "VISUAL",
>           "shape": [
>             3,
>             256,
>             256
>           ]
>         },
>         "observation.images.wrist_image": {
>           "type": "VISUAL",
>           "shape": [
>             3,
>             256,
>             256
>           ]
>         }
>       },
>       "max_action_dim": 32,
[... 389 lines cut ...]
'device': 'cuda',
'use_amp': False,
'steps': 30000,
'log_freq': 10
```

Transformers version: **4.53.0**

### 📉 Loss Output Sample

Here’s a snippet of the logs:

```
step:10  loss:0.160
step:20  loss:0.176
step:50  loss:0.176
step:100 loss:0.126
step:150 loss:0.127
step:200 loss:0.108
step:250 loss:0.098
step:300 loss:0.095
step:320 loss:0.101
```

Any insight or clarification would be appreciated! Thanks 🙏

hamondyan, 2025-08-21:
> <img alt="Image" width="332" [...]

Hello,

I am trying to train with the LIBERO dataset, but I encountered a problem: at the beginning of training, the loss is extremely high (around 500).

Could you confirm whether you are using the “IPEC-COMMUNITY/libero_spatial_no_noops_1.0.0_lerobot” dataset（here is the link to dataset：https://huggingface.co/datasets/IPEC-COMMUNITY/libero_spatial_no_noops_1.0.0_lerobot）?

Have you applied any preprocessing steps to the dataset? For example, converting actions to relative actions instead of absolute ones?

Any clarification on how you used the LIBERO dataset would be very helpful. Thanks!

pkooij, 2025-10-08:
Hi we update our pi0 implementation and released pi05 support in this PR: https://github.com/huggingface/lerobot/pull/1910 🚀 , and reproduced the pi05 libero results. If you still have this issue please reopen this thread, but I am closing it for now.

closed 2025-10-08 by pkooij