huggingface/lerobot #955: PI0FAST Evaluation results on LIBERO is very low

DelinQu, 2025-04-08:
Hi! Did anyone validate the performance of pi0-fast? I have trained pi0fast on the Libero dataset, but the deployment success rate is quite low. 
- training
[image]

- evaluation

```python
Task: pick up the black bowl on the wooden cabinet and place it on the plate
INFO:root:Starting episode 9...
INFO:root:Success: False
INFO:root:# episodes completed so far: 99
INFO:root:# successes: 3 (3.0%)
INFO:root:
Task: pick up the black bowl on the wooden cabinet and place it on the plate
INFO:root:Starting episode 10...
INFO:root:Success: False
INFO:root:# episodes completed so far: 100
INFO:root:# successes: 3 (3.0%)
INFO:root:Current task success rate: 0.0
INFO:root:Current total success rate: 0.03
INFO:root:Total success rate: 0.03
INFO:root:Total episodes: 100
INFO:OpenGL.platform.ctypesloader:Failed to load library ( 'libGLU.so.0' ): libGLU.so.0: cannot open shared object file: No such file or directory
```
https://github.com/user-attachments/assets/9977b9d5-b29b-4b8b-add7-2cff71f2cd71

- Instead, the jax version of pi0fast converges slowly but achieves ~95% Acc the libero-spatial benchmark. The Lerobot version pi0fast might counter the overfitting risk?

[image]

SahilChande, 2025-04-16:
Hie @DelinQu , I also want to evaluate the performace of pi0fast policy, But i dont know how to incorporate this policy in this lerobot for my so100 arm. If you can suggest me in this, i can evaluate performace.

ybu-lxd, 2025-04-27:
Hello, sorry to bother you. Which is the test environment for LIBERO? I can't find it in LeRobot.