huggingface/lerobot #956: pi0 multi gps train

ximiluuuu, 2025-04-08:
if i have multi 4090, how to modify to train pi0?

only 1 4090 just error
[image]

BruceGeLi, 2025-04-08:
same question + 1

ximiluuuu, 2025-04-09:
> same question + 1

sad

YanJiaHuan, 2025-04-09:
Also like to know.
Another question is how to freeze siglip and gemma, and only update the para for gemma expert?

YanJiaHuan, 2025-04-23:
they seems to freeze already, only train action expert, but it still taks up 29G on GPU.

ximiluuuu, 2025-04-23:
> they seems to freeze already, only train action expert, but it still taks up 29G on GPU.他们似乎已经冻结了，只训练 Action Expert，但它在 GPU 上仍然占用 29G。

sad

chenkang455, 2025-04-26:
> Also like to know. Another question is how to freeze siglip and gemma, and only update the para for gemma expert?

Try setting ``train_expert_only = True``, this will enable you to train your model with a single GPU with 24 G.

zttiannnn, 2025-06-23:
前辈们成功使用多卡训练了吗？😟