huggingface/lerobot #959: _get_query_indices function gives an error when LeRobotDataset initialized with episodes

ozgraslan, 2025-04-09:
### System Info

```Shell
- `lerobot` version: 0.1.0
- Platform: Linux-5.15.0-101-generic-x86_64-with-glibc2.35
- Python version: 3.10.4
- Huggingface_hub version: 0.30.1
- Dataset version: 3.4.1
- Numpy version: 1.26.4
- PyTorch version (GPU?): 2.6.0+cu124 (True)
- Cuda version: 12040
- Using GPU in script?: No
```

### Information

- [x] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

```python

from torch.utils.data import DataLoader
from lerobot.common.datasets.lerobot_dataset import LeRobotDataset, LeRobotDatasetMetadata

repo_id = "lerobot/libero_10_image"
ds_meta = LeRobotDatasetMetadata(repo_id)

print(f"Total number of episodes: {ds_meta.total_episodes}")
episodes = list(range(ds_meta.total_episodes))
test_episodes = episodes[-2:]
print(f"Test episode numbers: {test_episodes}")

delta_timestamps = {
    "action": [t / ds_meta.fps for t in range(64)],
}

dataset = LeRobotDataset(repo_id, episodes=test_episodes, delta_timestamps=delta_timestamps)

dataloader = DataLoader(
    dataset,
    num_workers=0,
    batch_size=32,
    shuffle=False,
)

for batch in dataloader:
    pass

```

### Expected behavior

When episodes are given in dataset initialization _get_query_indices function gives the following error:
```
    query_indices, padding = self._get_query_indices(idx, ep_idx)
  File "/lerobot/lerobot/common/datasets/lerobot_dataset.py", line 666, in _get_query_indices
    ep_start = self.episode_data_index["from"][ep_idx]
IndexError: index 377 is out of bounds for dimension 0 with size 2
```
This error happens due to ep_idx is the episode number in the full dataset, and get_episode_data_index returns an array with size=len(episodes). 

Correctly mapping the ep_idx to array idx with a dictionary can solve the issue. 
Adding to  __init__ function in lerobot_dataset.py [after line 505](https://github.com/huggingface/lerobot/blob/2c86fea78aa3a3177411a9758a0a703baac74ea8/lerobot/common/datasets/lerobot_dataset.py#L505): 
```python 
self.ep_idx_to_arr_idx = {ep_idx: arr_idx for arr_idx, ep_idx in enumerate(episodes)}
```
Changing the [line 730](https://github.com/huggingface/lerobot/blob/2c86fea78aa3a3177411a9758a0a703baac74ea8/lerobot/common/datasets/lerobot_dataset.py#L730) in lerobot_dataset.py __getitem__ function: 
```python
query_indices, padding = self._get_query_indices(idx, self.ep_idx_to_arr_idx[ep_idx])
```
solves the issue.

CarolinePascal, 2025-04-14:
Hi @ozgraslan !

Thank you for suggesting a solution to this issue. Could you possibly fill a PR so that your fix may be merged into `main` ? 

Thanks,

Caroline.

jfelip, 2025-10-08:
This is still an issue in lerobot version 0.3.3 (and apparently in the main branch too)

The solution from @ozgraslan works. Still I think it will need to check if episodes is None and create a full 1 to 1 mapping for the episode indices if that is the case.

At the end of the __init__ method in lerobot_dataset.py
```python
        # Setup episode index to array index mapping
        if episodes is not None:
            self.ep_idx_to_arr_idx = {ep_idx: arr_idx for arr_idx, ep_idx in enumerate(episodes)}
        else:
            self.ep_idx_to_arr_idx = {ep_idx: ep_idx for ep_idx in range(len(self.episode_data_index["from"]))}
```

And in the getitem method
```diff
    def __getitem__(self, idx) -> dict:
        item = self.hf_dataset[idx]
        ep_idx = item["episode_index"].item()

        query_indices = None
        if self.delta_indices is not None:
+          query_indices, padding = self._get_query_indices(idx, self.ep_idx_to_arr_idx[ep_idx])
-          query_indices, padding = self._get_query_indices(idx, ep_idx)
            query_result = self._query_hf_dataset(query_indices)
            item = {**item, **padding}
            for key, val in query_result.items():
                item[key] = val

        if len(self.meta.video_keys) > 0:
            current_ts = item["timestamp"].item()
            query_timestamps = self._get_query_timestamps(current_ts, query_indices)
+          video_frames = self._query_videos(query_timestamps, self.ep_idx_to_arr_idx[ep_idx])
-          video_frames = self._query_videos(query_timestamps, ep_idx)
            item = {**video_frames, **item}

        if self.image_transforms is not None:
            image_keys = self.meta.camera_keys
            for cam in image_keys:
                item[cam] = self.image_transforms(item[cam])

        # Add task as a string
        task_idx = item["task_index"].item()
        item["task"] = self.meta.tasks[task_idx]

        return item
```

ozgraslan, 2025-10-20:
Hello again, 
Seems like with the current version the bug is fixed. I did not get the same error with `lerobot/pusht`. 
I was not able to test it with 'lerobot/libero_10_image', because I am getting the following error:

```
Traceback (most recent call last):
  File "/home/git/mylerobot/test_bug.py", line 6, in <module>
    ds_meta = LeRobotDatasetMetadata(repo_id)
  File "/home/git/mylerobot/src/lerobot/datasets/lerobot_dataset.py", line 102, in __init__
    self.load_metadata()
  File "/home/git/mylerobot/src/lerobot/datasets/lerobot_dataset.py", line 161, in load_metadata
    check_version_compatibility(self.repo_id, self._version, CODEBASE_VERSION)
  File "/home/git/mylerobot/src/lerobot/datasets/utils.py", line 469, in check_version_compatibility
    raise BackwardCompatibilityError(repo_id, v_check)
  File "/home/git/mylerobot/src/lerobot/datasets/backward_compatibility.py", line 47, in __init__
    raise NotImplementedError(
NotImplementedError: Contact the maintainer on [Discord](https://discord.com/invite/s3KuuzsPFb).
```

hairuoliu1, 2026-02-03:
It works for me.