huggingface/lerobot #963: unrecognized arguments: --local-files-only 1

milong26, 2025-04-09:
### System Info

```Shell
I try to [visualize datase](https://github.com/huggingface/lerobot#visualize-datasets)  
As I have downloaded dataset locally, I use this command:

python lerobot/scripts/visualize_dataset.py \
    --repo-id lerobot/berkeley_autolab_ur5 \
    --root /path \
    --local-files-only 1 \
    --episode-index 0

Then it said this:

usage: visualize_dataset.py [-h] --repo-id REPO_ID --episode-index EPISODE_INDEX [--root ROOT] [--output-dir OUTPUT_DIR] [--batch-size BATCH_SIZE]
                            [--num-workers NUM_WORKERS] [--mode MODE] [--web-port WEB_PORT] [--ws-port WS_PORT] [--save SAVE]
                            [--tolerance-s TOLERANCE_S]
visualize_dataset.py: error: unrecognized arguments: --local-files-only 1

How to fix command?
```

### Information

- [ ] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

install LeRobot and try to visualize

### Expected behavior

I can visualize through local dataset,since downloading large data from huggingface is difficult

milong26, 2025-04-10:
ok just using --local is fine. LIke this:

I download my dataset in `~/Downloads/pusht` (I use git to download huggingface dataset at ~/Downloads)

So I use this command `python lerobot/scripts/visualize_dataset.py     --repo-id lerobot/pusht     --episode-index 0 --root ~/Downloads/pusht/`

If anyone need help at this.

And please correct readme.md

milong26, 2025-04-10:
By the way I found the data in `lerobot/berkeley_autolab_ur5` is strange, I can't understand why its observation/state is 8-dimensional ;-; What does joint0/1/2/3... in lerobot mean?

joint pos or just xyz+rxryrzrw?

milong26, 2025-04-29:
However in `visualize_html` i can't visualize a local dataset even with 'local', it just gets dataset from hub

milong26, 2025-04-30:
> However in `visualize_html` i can't visualize a local dataset even with 'local', it just gets dataset from hub

So to visualize local dataset I still use `python lerobot/scripts/visualize_dataset.py     --repo-id lerobot/pusht     --episode-index 0 --root LOCAL_PATH(like ~/.cache/huggingface/lerobot/.../)` while trpo-id just decorate :) this works

tc-huang, 2025-04-30:
Hello @milong26,

I created PR #1057 about using `--repo-id` and `--root` instead of the old `--local-files-only` argument for visualizing data with `lerobot/scripts/visualize_dataset.py` and `lerobot/scripts/visualize_dataset_html.py`.

It might be relevant to your case, so feel free to take a look at PR #1057 if you're interested.

closed 2025-05-05 by milong26