huggingface/lerobot #970: ValueError: No valid stream found in input file. Is -1 of the desired media type?

taiji1985, 2025-04-11:
### System Info

```Shell
lerobot 0.1.0  , Ubuntu 24.04 ,numpy 2.2.4 ,torch '2.6.0+cu124'

(lerobotv2) yang@yang-gpu:~/new/lerobot$ python lerobot/scripts/train.py     --policy.type=act     --dataset.repo_id=lerobot/aloha_sim_insertion_human     --env.type=aloha     --output_dir=outputs/train/act_aloha_insertion
INFO 2025-04-11 21:33:38 ils/utils.py:44 Cuda backend detected, using cuda.
WARNING 2025-04-11 21:33:38 /policies.py:67 Device 'None' is not available. Switching to 'cuda'.
INFO 2025-04-11 21:33:38 ts/train.py:111 {'batch_size': 8,
 'dataset': {'episodes': None,
             'image_transforms': {'enable': False,
                                  'max_num_transforms': 3,
                                  'random_order': False,
                                  'tfs': {'brightness': {'kwargs': {'brightness': [0.8,
                                                                                   1.2]},
                                                         'type': 'ColorJitter',
                                                         'weight': 1.0},
                                          'contrast': {'kwargs': {'contrast': [0.8,
                                                                               1.2]},
                                                       'type': 'ColorJitter',
                                                       'weight': 1.0},
                                          'hue': {'kwargs': {'hue': [-0.05,
                                                                     0.05]},
                                                  'type': 'ColorJitter',
                                                  'weight': 1.0},
                                          'saturation': {'kwargs': {'saturation': [0.5,
                                                                                   1.5]},
                                                         'type': 'ColorJitter',
                                                         'weight': 1.0},
                                          'sharpness': {'kwargs': {'sharpness': [0.5,
                                                                                 1.5]},
                                                        'type': 'SharpnessJitter',
                                                        'weight': 1.0}}},
             'repo_id': 'lerobot/aloha_sim_insertion_human',
             'revision': None,
             'root': None,
             'use_imagenet_stats': True,
             'video_backend': 'torchcodec'},
 'env': {'episode_length': 400,
         'features': {'action': {'shape': [14],
                                 'type': <FeatureType.ACTION: 'ACTION'>},
                      'agent_pos': {'shape': [14],
                                    'type': <FeatureType.STATE: 'STATE'>},
                      'pixels/top': {'shape': [480, 640, 3],
                                     'type': <FeatureType.VISUAL: 'VISUAL'>}},
         'features_map': {'action': 'action',
                          'agent_pos': 'observation.state',
                          'pixels/top': 'observation.images.top',
                          'top': 'observation.image.top'},
         'fps': 50,
         'obs_type': 'pixels_agent_pos',
         'render_mode': 'rgb_array',
         'task': 'AlohaInsertion-v0',
         'type': 'aloha'},
 'eval': {'batch_size': 50, 'n_episodes': 50, 'use_async_envs': False},
 'eval_freq': 20000,
 'job_name': 'aloha_act',
 'log_freq': 200,
 'num_workers': 4,
 'optimizer': {'betas': [0.9, 0.999],
               'eps': 1e-08,
               'grad_clip_norm': 10.0,
               'lr': 1e-05,
               'type': 'adamw',
               'weight_decay': 0.0001},
 'output_dir': 'outputs/train/act_aloha_insertion',
 'policy': {'chunk_size': 100,
            'device': 'cuda',
            'dim_feedforward': 3200,
            'dim_model': 512,
[... 213 lines cut ...]
  libavutil      56. 51.100 / 56. 51.100
  libavcodec     58. 91.100 / 58. 91.100
  libavformat    58. 45.100 / 58. 45.100
  libavdevice    58. 10.100 / 58. 10.100
  libavfilter     7. 85.100 /  7. 85.100
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  7.100 /  5.  7.100
  libswresample   3.  7.100 /  3.  7.100
  libpostproc    55.  7.100 / 55.  7.100
Hyper fast Audio and Video encoder
usage: ffmpeg [options] [[infile options] -i infile]... {[outfile options] outfile}...
```

### Information

- [ ] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

python lerobot/scripts/train.py     --policy.type=act     --dataset.repo_id=lerobot/aloha_sim_insertion_human     --env.type=aloha     --output_dir=outputs/train/act_aloha_insertion

### Expected behavior

train start

karpenet, 2025-04-11:
Following

zymsun2003, 2025-04-12:
Modify the code:
Line.35:  ./lerobot/common/datasets/video_utils.py
return 'torchcodec' --> return 'pyav'

bearpaw, 2025-04-14:
Same issue here. I followed the instruction to install the conda env. 
```bash
python lerobot/scripts/visualize_dataset.py \
    --repo-id lerobot/pusht \
    --episode-index 0
```
Output
```bash
Resolving data files: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████| 206/206 [00:00<00:00, 534009.04it/s]
[2025-04-14T21:53:33Z INFO  winit::platform_impl::linux::x11::window] Guessed window scale factor: 1
  0%|                                                                                                                                            | 0/6 [00:00<?, ?it/s]
Traceback (most recent call last):
  File "/home/weiy/Projects/lerobot/lerobot/scripts/visualize_dataset.py", line 292, in <module>
    main()
  File "/home/weiy/Projects/lerobot/lerobot/scripts/visualize_dataset.py", line 288, in main
    visualize_dataset(dataset, **vars(args))
  File "/home/weiy/Projects/lerobot/lerobot/scripts/visualize_dataset.py", line 147, in visualize_dataset
    for batch in tqdm.tqdm(dataloader, total=len(dataloader)):
  File "/mnt/ssd/miniconda3/envs/lerobot/lib/python3.10/site-packages/tqdm/std.py", line 1181, in __iter__
    for obj in iterable:
  File "/mnt/ssd/miniconda3/envs/lerobot/lib/python3.10/site-packages/torch/utils/data/dataloader.py", line 708, in __next__
    data = self._next_data()
  File "/mnt/ssd/miniconda3/envs/lerobot/lib/python3.10/site-packages/torch/utils/data/dataloader.py", line 1480, in _next_data
    return self._process_data(data)
  File "/mnt/ssd/miniconda3/envs/lerobot/lib/python3.10/site-packages/torch/utils/data/dataloader.py", line 1505, in _process_data
    data.reraise()
  File "/mnt/ssd/miniconda3/envs/lerobot/lib/python3.10/site-packages/torch/_utils.py", line 733, in reraise
    raise exception
ValueError: Caught ValueError in DataLoader worker process 0.
Original Traceback (most recent call last):
  File "/mnt/ssd/miniconda3/envs/lerobot/lib/python3.10/site-packages/torch/utils/data/_utils/worker.py", line 349, in _worker_loop
    data = fetcher.fetch(index)  # type: ignore[possibly-undefined]
  File "/mnt/ssd/miniconda3/envs/lerobot/lib/python3.10/site-packages/torch/utils/data/_utils/fetch.py", line 52, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/mnt/ssd/miniconda3/envs/lerobot/lib/python3.10/site-packages/torch/utils/data/_utils/fetch.py", line 52, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/weiy/Projects/lerobot/lerobot/common/datasets/lerobot_dataset.py", line 739, in __getitem__
    video_frames = self._query_videos(query_timestamps, ep_idx)
  File "/home/weiy/Projects/lerobot/lerobot/common/datasets/lerobot_dataset.py", line 711, in _query_videos
    frames = decode_video_frames(video_path, query_ts, self.tolerance_s, self.video_backend)
  File "/home/weiy/Projects/lerobot/lerobot/common/datasets/video_utils.py", line 66, in decode_video_frames
    return decode_video_frames_torchcodec(video_path, timestamps, tolerance_s)
  File "/home/weiy/Projects/lerobot/lerobot/common/datasets/video_utils.py", line 195, in decode_video_frames_torchcodec
    decoder = VideoDecoder(video_path, device=device, seek_mode="approximate")
  File "/mnt/ssd/miniconda3/envs/lerobot/lib/python3.10/site-packages/torchcodec/decoders/_video_decoder.py", line 98, in __init__
    core.add_video_stream(
  File "/mnt/ssd/miniconda3/envs/lerobot/lib/python3.10/site-packages/torch/_ops.py", line 723, in __call__
    return self._op(*args, **kwargs)
ValueError: No valid stream found in input file. Is -1 of the desired media type?
```
My OS is Ubuntu 24.04.1

taiji1985, 2025-04-15:
> Same issue here. I followed the instruction to install the conda env.
> 
> python lerobot/scripts/visualize_dataset.py \
>     --repo-id lerobot/pusht \
>     --episode-index 0
> Output
> 
> Resolving data files: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████| 206/206 [00:00<00:00, 534009.04it/s]
> [2025-04-14T21:53:33Z INFO  winit::platform_impl::linux::x11::window] Guessed window scale factor: 1
>   0%|                                                                                                                                            | 0/6 [00:00<?, ?it/s]
> Traceback (most recent call last):
>   File "/home/weiy/Projects/lerobot/lerobot/scripts/visualize_dataset.py", line 292, in <module>
>     main()
>   File "/home/weiy/Projects/lerobot/lerobot/scripts/visualize_dataset.py", line 288, in main
>     visualize_dataset(dataset, **vars(args))
>   File "/home/weiy/Projects/lerobot/lerobot/scripts/visualize_dataset.py", line 147, in visualize_dataset
>     for batch in tqdm.tqdm(dataloader, total=len(dataloader)):
>   File "/mnt/ssd/miniconda3/envs/lerobot/lib/python3.10/site-packages/tqdm/std.py", line 1181, in __iter__
>     for obj in iterable:
>   File "/mnt/ssd/miniconda3/envs/lerobot/lib/python3.10/site-packages/torch/utils/data/dataloader.py", line 708, in __next__
>     data = self._next_data()
>   File "/mnt/ssd/miniconda3/envs/lerobot/lib/python3.10/site-packages/torch/utils/data/dataloader.py", line 1480, in _next_data
>     return self._process_data(data)
>   File "/mnt/ssd/miniconda3/envs/lerobot/lib/python3.10/site-packages/torch/utils/data/dataloader.py", line 1505, in _process_data
>     data.reraise()
>   File "/mnt/ssd/miniconda3/envs/lerobot/lib/python3.10/site-packages/torch/_utils.py", line 733, in reraise
>     raise exception
> ValueError: Caught ValueError in DataLoader worker process 0.
> Original Traceback (most recent call last):
>   File "/mnt/ssd/miniconda3/envs/lerobot/lib/python3.10/site-packages/torch/utils/data/_utils/worker.py", line 349, in _worker_loop
>     data = fetcher.fetch(index)  # type: ignore[possibly-undefined]
>   File "/mnt/ssd/miniconda3/envs/lerobot/lib/python3.10/site-packages/torch/utils/data/_utils/fetch.py", line 52, in fetch
>     data = [self.dataset[idx] for idx in possibly_batched_index]
>   File "/mnt/ssd/miniconda3/envs/lerobot/lib/python3.10/site-packages/torch/utils/data/_utils/fetch.py", line 52, in <listcomp>
>     data = [self.dataset[idx] for idx in possibly_batched_index]
>   File "/home/weiy/Projects/lerobot/lerobot/common/datasets/lerobot_dataset.py", line 739, in __getitem__
>     video_frames = self._query_videos(query_timestamps, ep_idx)
>   File "/home/weiy/Projects/lerobot/lerobot/common/datasets/lerobot_dataset.py", line 711, in _query_videos
>     frames = decode_video_frames(video_path, query_ts, self.tolerance_s, self.video_backend)
>   File "/home/weiy/Projects/lerobot/lerobot/common/datasets/video_utils.py", line 66, in decode_video_frames
>     return decode_video_frames_torchcodec(video_path, timestamps, tolerance_s)
>   File "/home/weiy/Projects/lerobot/lerobot/common/datasets/video_utils.py", line 195, in decode_video_frames_torchcodec
>     decoder = VideoDecoder(video_path, device=device, seek_mode="approximate")
>   File "/mnt/ssd/miniconda3/envs/lerobot/lib/python3.10/site-packages/torchcodec/decoders/_video_decoder.py", line 98, in __init__
>     core.add_video_stream(
>   File "/mnt/ssd/miniconda3/envs/lerobot/lib/python3.10/site-packages/torch/_ops.py", line 723, in __call__
>     return self._op(*args, **kwargs)
> ValueError: No valid stream found in input file. Is -1 of the desired media type?
> My OS is Ubuntu 24.04.1

use this to  reinstall  ffmpeg (or you can remove it first )

conda install -c conda-forge  ffmpeg=6.1.1 -y

bhwxzx, 2025-05-09:
dataset = LeRobotDataset(repo_id, root=root_path, video_backend="pyav")

bhwxzx, 2025-05-09:
dataset = LeRobotDataset(repo_id, root=root_path, video_backend="pyav") you can load dataset with video_backend="pyav"

Spring-Lee-cpu, 2025-06-12:
> Modify the code:  修改代码： Line.35: ./lerobot/common/datasets/video_utils.py第 35 行：./lerobot/common/datasets/video_utils.py return 'torchcodec' --> return 'pyav'返回 'torchcodec' --> 返回 'pyav'

This works for me! I have another question, where is the downloaded data stored?