huggingface/lerobot #973: Error when using the manual calibration

PhGoffin, 2025-04-12:
### System Info

```Shell
- `lerobot` version: 0.1.0
- Platform: Windows-10-10.0.26100-SP0
- Python version: 3.10.16
- Huggingface_hub version: 0.30.2
- Dataset version: 3.5.0
- Numpy version: 2.2.4
- PyTorch version (GPU?): 2.6.0+cpu (False)
- Cuda version: N/A
- Using GPU in script?: <fill in>

Computer Geekom A7
Processeur	AMD Ryzen 9 7940HS w/ Radeon 780M Graphics, 4001 MHz, 8 cores
Using Anaconda
```

### Information

- [x] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

Dear,
I just finalized the 3D Print of the robots (leader/follower). I followed the instructions from the site: https://github.com/huggingface/lerobot/blob/main/examples/10_use_so100.md
So, I am really a newbie ! the bug is for sure on my side, nothing to do with your code!

I detected the USB Port COM12 & COM13, updated the class " So100RobotConfig"
Of course, all my motors are assigned with an ID.
I have no camera connected on the robot.

When I tried to run the manual calibration, I got an error (miss configuration?)

Thanks for your support, I am lost!
  Philippe

python lerobot/scripts/control_robot.py --robot.type=so100 --robot.cameras='{}' --control.type=calibrate  --control.arms='["main_follower"]'
or
python lerobot/scripts/control_robot.py --robot.type=so100 --robot.cameras='{}' --control.type=calibrate  --control.arms='["main_leader"]'

### Expected behavior

This is the error received:

(lerobot) C:\Users\artco\~\lerobot>python lerobot/scripts/control_robot.py --robot.type=so100 --robot.cameras='{}' --control.type=calibrate  --control.arms='["main_follower"]'
Traceback (most recent call last):
  File "C:\Users\artco\miniconda3\envs\lerobot\lib\site-packages\draccus\parsers\decoding.py", line 133, in decode_dataclass
    field_value = get_decoding_fn(field_type)(raw_value, (*path, name))
  File "C:\Users\artco\miniconda3\envs\lerobot\lib\site-packages\draccus\parsers\decoding.py", line 463, in _decode_dict
    items = raw_value.items()
AttributeError: 'str' object has no attribute 'items'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Users\artco\~\lerobot\lerobot\scripts\control_robot.py", line 394, in <module>
    control_robot()
  File "C:\Users\artco\~\lerobot\lerobot\configs\parser.py", line 226, in wrapper_inner
    cfg = draccus.parse(config_class=argtype, config_path=config_path, args=cli_args)
  File "C:\Users\artco\miniconda3\envs\lerobot\lib\site-packages\draccus\argparsing.py", line 211, in parse
    return parser.parse_args(args)
  File "C:\Users\artco\miniconda3\envs\lerobot\lib\site-packages\draccus\argparsing.py", line 102, in parse_args
    args, _ = self.parse_known_args(args, namespace, is_parse_args=True)
  File "C:\Users\artco\miniconda3\envs\lerobot\lib\site-packages\draccus\argparsing.py", line 136, in parse_known_args
    parsed_t = self._postprocessing(parsed_args)
  File "C:\Users\artco\miniconda3\envs\lerobot\lib\site-packages\draccus\argparsing.py", line 180, in _postprocessing
    cfg = decoding.decode(self.config_class, deflat_d)
  File "C:\Users\artco\miniconda3\envs\lerobot\lib\site-packages\draccus\parsers\registry_utils.py", line 78, in wrapper
    return base_func(*args, **kw)
  File "C:\Users\artco\miniconda3\envs\lerobot\lib\site-packages\draccus\parsers\decoding.py", line 48, in decode
    return get_decoding_fn(cls)(raw_value, ())  # type: ignore
  File "C:\Users\artco\miniconda3\envs\lerobot\lib\site-packages\draccus\parsers\decoding.py", line 135, in decode_dataclass
    raise e
  File "C:\Users\artco\miniconda3\envs\lerobot\lib\site-packages\draccus\parsers\decoding.py", line 133, in decode_dataclass
    field_value = get_decoding_fn(field_type)(raw_value, (*path, name))
  File "C:\Users\artco\miniconda3\envs\lerobot\lib\site-packages\draccus\parsers\decoding.py", line 201, in decode_choice_class
    return decode_dataclass(subcls, raw_value, path)
  File "C:\Users\artco\miniconda3\envs\lerobot\lib\site-packages\draccus\parsers\decoding.py", line 137, in decode_dataclass
    raise DecodingError(
draccus.utils.DecodingError: `robot.cameras`: Failed when parsing value=''{}'' into field "<class 'lerobot.common.robot_devices.robots.configs.So100RobotConfig'>.cameras" of type dict[str, lerobot.common.robot_devices.cameras.configs.CameraConfig].
        Underlying error is "AttributeError: 'str' object has no attribute 'items'"

imstevenpmwork, 2025-04-12:
From the logs you shared it seems you are not currently in the latest main, I recommend you updating your local 😄

jlhausherr, 2025-04-21:
Hi, I've just experienced the same situation.
Under Windows, double quote  " are misinterpreted.
In script control_robot.py I found this script which works, but the two arms must be connected. python _lerobot/scripts/control_robot.py --robot.type=so100 --control.type=calibrate_
 Hope this helps.