huggingface/lerobot #981: How can I simulate robots without physical robots? How should I learn simulation robots? Do you have any good recommendations?

harryhu0301, 2025-04-15:
How can I simulate robots without physical robots? How should I learn simulation robots? Do you have any good recommendations?I am a beginner.

stefangolas, 2025-04-16:
Check out this repo https://huggingface.co/lerobot/diffusion_pusht and note the env.type argument. That specifies a simulation environment.

I'm working on a ROS integration with Isaac Sim so you can simulate using the normal robot control scripts but I dont think there's any existing way right now.

drf63329272, 2025-04-28:
@stefangolas Hi , "https://huggingface.co/lerobot/diffusion_pusht",this is not repo addr.where is this repo ?

gokulp01, 2025-06-05:
@drf63329272 @harryhu0301 here are some good resources:  
https://github.com/TheRobotStudio/SO-ARM100/tree/main/Simulation
https://wiki.seeedstudio.com/lerobot_so100m_isaacsim/
https://github.com/michel-aractingi/gym-lowcostrobot/
https://github.com/huggingface/gym-pusht

16Huzeyu, 2025-07-15:
@drf63329272 @harryhu0301 https://github.com/LightwheelAI/leisaac

jadechoghari, 2025-10-17:
We also provide a Imitaiton learning with Sim example: https://huggingface.co/docs/lerobot/en/il_sim